apache/nuttx #16265: [HELP] Why is clock_nanosleep not actually high precision?

dejavudwh, 2025-04-25:
### Description

From what I can see in the implementation, clock_nanosleep enters sleep via wd_start_abstick and is later woken up in wd_timer. However, wd_timer is triggered by the same timer that handles the OS tick. This means the actual precision of clock_nanosleep depends on the OS tick interval.

Is my understanding correct? If so, I’m wondering why this approach was chosen.

Why not use a separate timer to handle the wake-up, or, as in Linux, repurpose the timer originally used for the OS tick to be managed by hrtimer, and then emulate the OS tick based on that? Both approaches could help achieve higher precision to some extent.

I might be missing something—please correct me if I’m wrong.

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-04-27:
@xiaoxiang781216 @anchao @raiden00pl ^

@dejavudwh I think not all microcontrollers have hrtimer. I know some families of STM32 has support to it.

xiaoxiang781216, 2025-04-27:
All nuttx timeout use tick unit, so you need just CONFIG_USEC_PER_TICK to match your precision.
But please note that you have to switch tickless mode to reduce the timer interrupt overhead if the precision is around ~10us or ~100us.

dejavudwh, 2025-04-27:
> [@xiaoxiang781216](https://github.com/xiaoxiang781216) [@anchao](https://github.com/anchao) [@raiden00pl](https://github.com/raiden00pl) ^
> 
> [@dejavudwh](https://github.com/dejavudwh) I think not all microcontrollers have hrtimer. I know some families of STM32 has support to it.

You are correct. However, the "hrtimer" I mentioned here is a software concept similar to Linux's hrtimer. It does not require hardware-specific "hrtimer" support, but simply enables shorter-duration timing. For example, with a 24M timer, it can already achieve ~40ns-level timing precision. Currently, NuttX's minimum timing precision remains one OS Tick.

dejavudwh, 2025-04-27:
> All nuttx timeout use tick unit, so you need just CONFIG_USEC_PER_TICK to match your precision.
> But please note that you have to switch tickless mode to reduce the timer interrupt overhead if the precision is around ~10us or ~100us.

@xiaoxiang781216 Thank you for your answer. This is something I’m curious about: Why did NuttX choose this design approach? Since there are use cases requiring both periodic OS Ticks and higher-precision timers, Linux addresses this by using hrtimer to provide high-precision timing while emulating a tick device within hrtimer to maintain periodic OS ticks.

xiaoxiang781216, 2025-04-27:
Linux support two set of timer:

1. timer(https://www.kernel.org/doc/html/v4.9/driver-api/basics.html#c.add_timer) use tick unit
2. hrtimer(https://docs.kernel.org/timers/hrtimers.html) use ns unit

But NuttX just support wdog(similar Linux timer), no hrtimer. The problem is NuttX need develop the similar facility.

dejavudwh, 2025-04-28:
> Linux support two set of timer:
>
> 1. timer(https://www.kernel.org/doc/html/v4.9/driver-api/basics.html#c.add_timer) use tick unit
> 2. hrtimer(https://docs.kernel.org/timers/hrtimers.html) use ns unit [...]

@xiaoxiang781216 Understood. Thank you very much for your answer.