apache/nuttx #16269: [HELP] <title> Please explain the esp32 wlan driver working which cause the following issue on cpu0

abisop, 2025-04-26:
### Description

Device -  ESP32-s3 
config - esp32s3-devkit:wifi

using the command "wapi scan wlan0" fails resulting in cpu reset (most of the time)

and using wapi pscan wlan0 results folowing

[image]

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-04-27:
@tmedicci could you please take a look?

tmedicci, 2025-04-28:
Hi @abisop , thanks for reporting.

I tried it locally and couldn't reproduce the issue:
```
nsh> wapi scan wlan0
bssid / frequency / signal level / encode / ssid
xx:xx:xx:xx:xx:xx	2462	-76	0800	xxxx
xx:xx:xx:xx:xx:xx	2437	-70	0800	xxxx
xx:xx:xx:xx:xx:xx	2412	-64	0800	xxxx
xx:xx:xx:xx:xx:xx	2412	-64	0800	xxxx
xx:xx:xx:xx:xx:xx	2447	-58	0800	xxxx
xx:xx:xx:xx:xx:xx	2412	-29	0800	xxxx
xx:xx:xx:xx:xx:xx	2412	-29	0800	xxxx
xx:xx:xx:xx:xx:xx	2412	-28	0800	xxxx
xx:xx:xx:xx:xx:xx	2412	-28	0800	xxxx
```

Can you please provide more information about the issue? Specifically, running `make host_info` with your ESP32-S3 device attached would be very helpful.

abisop, 2025-04-29:
[image]
[host_info.txt](https://github.com/user-attachments/files/19965625/host_info.txt)
@tmedicci Here is the host info 

Also I observed that after powering device the error resolves after some time(10-15seconds).

Maybe some kind of initialization taking time.

but after that everything is fine.

acassis, 2025-04-29:
@abisop could you add more info? If the issue happens in the first 10s it still needs to be fixed/investigated.

abisop, 2025-04-30:
@tmedicci 
I found when turning on some extra configurations like (CONFIG_FS_ROMFS, CONFIG_DEBUG_FEATURES ) along with the following
> Device - ESP32-s3 config - esp32s3-devkit:wifi
The issue occurs...

Please give me some idea about how/where to inspect...

What part of software can cause software reset...?

abisop, 2025-05-02:
@acassis @tmedicci For test purpose I hooked up printf in function esp_wifi_start_scan( nuttx/arch/xtensa/src/common/espressif/esp32_wifi_utils.c )
[image]
The following results can be seen : (active scan)
[image]

> The flow is interupted and esp restarts

while for passive scan which uses same function i.e esp_wifi_start_scan it works fine :

[image]

fdcavalcanti, 2025-05-05:
According to your print screens, it is resetting on `GLITCH_RTC_RST`.

First of all, is it a custom board or a Espressif devkit?

Can you try a different USB port? This could be power related or something on board. See similar issue [here](https://github.com/Xinyuan-LilyGO/LilyGo-T-OI-PLUS/issues/3).

abisop, 2025-05-05:
Thanks @fdcavalcanti sir for responding 
I don't think its a power issue
As far as i got to know, its some kind of memory and timing (maybe race condition) issue...

Please see this - 
https://github.com/user-attachments/assets/e3d60979-1a9b-4abd-be22-3d8c9a9dcf9c

let me explain :
For lets say we execute Active scan :

1 ) WAPI send command to low level driver for scan.
2) It checks if the scan(or any other request) is complete 
[image]

3) The previous step is necessary to get the data from lower driver 

So if we ask him for data before it (lower driver) finishes. ---> (REBOOT/CRASH)

Also : 
>If test with esp32s3-devkit:wifi config no issue will be there but if we add some more features such as ROMFS, Telnettd, etc
>This happens 

Till now I am able to fix the issue by hardcoding a delay:
To be continued.....

fdcavalcanti, 2025-05-05:
Ok so it is not the wifi defconfig by itself, right?
Please send here your current defconfig so I can replicate with the actual features that are enabled (`make savedefconfig`).

fdcavalcanti, 2025-05-06:
I used your `defconfig` and failed to replicate. `wapi scan wlan0` worked normally right after a reboot. 

Also, I asked this before, but are you using a devkit or a custom board?
Can you list the exact steps taken to reproduce the problem?

abisop, 2025-05-06:
@fdcavalcanti Its a xiao esp32s3 -> https://wiki.seeedstudio.com/xiao_esp32s3_getting_started/

[image]

> Can you list the exact steps taken to reproduce the problem?
Can you add some more tasks similar to telnetd/etc_romfs 
and are you able to 
1) connect using " wapi essid " command 
2) renew wlan0
3) wapi save_config

fdcavalcanti, 2025-05-06:
I used your defconfig on my `ESP32S3-Devkit`. Telnet is up, ROMFS is selected and connected to a hotspot, did a wireless scan with `wapi scan wlan0`. No issues.

```
nsh> ls /etc/init.d
/etc/init.d:
 .
 ..
 rc.sysinit
 rcS
nsh> ps
  PID GROUP CPU PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK            STACK COMMAND
    0     0   0   0 FIFO     Kthread   - Assigned           0000000000000000 0003056 CPU0 IDLE
    1     0   1   0 FIFO     Kthread   - Running            0000000000000000 0003048 CPU1 IDLE
    2     0 --- 100 RR       Kthread   - Waiting  Semaphore 0000000000000000 0004032 lpwork 0x3fca4d30 0x3fca4d60
    3     3   0 100 RR       Task      - Running            0000000000000000 0008144 nsh_main
    4     0 --- 255 RR       Kthread   - Waiting  Semaphore 0000000000000000 0000704 spiflash_op 0x3fcac80c
    5     0 --- 255 RR       Kthread   - Waiting  Semaphore 0000000000000000 0000704 spiflash_op 0x3fcac80c
    6     0 --- 223 RR       Kthread   - Waiting  Semaphore 0000000000000000 0004048 rt_timer
    7     0 --- 253 RR       Kthread   - Waiting  MQ empty  0000000000000000 0006616 wifi
    8     8 --- 100 RR       Task      - Waiting  Semaphore 0000000000000000 0004056 telnetd
nsh> 
```

This could be board related (even power related, as I mentioned). There's a `esp32s3-xiao` available, however it seems to be in the initial stages. Seems it was added here #16161. Maybe you can contact the guys responsible.

Honestly I can't replicate without the proper board, and unfortunately I don't have the XIAO one.

abisop, 2025-05-07:
> Honestly I can't replicate without the proper board, and unfortunately I don't have the XIAO one.

As long I am working on something internal like wifi and its esp32 s3. I think variations are there only in memory size (Flash and PSRAM). So people who wrote common part of esp32s3 wrote linker in such a way so that it can be changed according to config

> Note : passive scan works fine , scan_results works fine , active scan (cause problem) but works sometimes 
So I invested time and found this problem....

> ...

> Please see this - https://github.com/user-attachments/assets/e3d60979-1a9b-4abd-be22-3d8c9a9dcf9c
> 
> let me explain : For lets say we execute Active scan :
> 
> 1 ) WAPI send command to low level driver for scan, connect, etc. 2) It checks if the scan(or any other request) is complete [image]
> 
>    3. The previous step is necessary to get the data from lower driver
> 
> 
> So if we ask him for data before it (lower driver) finishes. ---> (REBOOT/CRASH)
> 
> Also :
> 
> > If test with esp32s3-devkit:wifi config no issue will be there but if we add some more features such as ROMFS, Telnettd, etc
> > This happens
> 
> Till now I am able to fix the issue by hardcoding a delay: To be continued.....
>

fdcavalcanti, 2025-05-07:
Yes I understand it is the same chip, I tried your exact defconfig and could not replicate on mine.
I noticed you are using USB Serial, did you change that when you sent the defconfig?

abisop, 2025-05-14:
It resulted in a power issue, which caused device to reboot. 
I resolved it by decreasing the tx power...

acassis, 2025-05-14:
@abisop nice to let us know and we are glad you got it working. Please close this Issue

closed 2025-05-14 by abisop