apache/nuttx #16271: [FEATURE] raspberrypi pico w/2/2w firmware flashing automation with make flash

cederom, 2025-04-26:
### Is your feature request related to a problem? Please describe.

* For now in order to flash rPI-Pico(-W/-2/-2W) we need to manually press Boot button on USB attach then copy-paste firmware file.
* That blocks automation testing. 

### Describe the solution you'd like

* Use `make flash` to have firmware flashed and ready for runtime testing.
* Enable full automation of firmware flashing.

### Describe alternatives you've considered

@keever50 reported that `picotool` (something like `esptool` for ESP32) can be used to flash firmware but still requires manual Boot button interaction, and PlatformIO can switch to BootROM by some magic sequence. That may indicate flashing automation should be possible. 

### Verification

- [x] I have verified before submitting the report.

cederom, 2025-04-27:
@linguini1 as you are most familiar with rPI-Pico can we assign you this task to research in a free moment? :-) Having this single usb port flashing just as with esptool would enable automated testing on a real world hardware :-)

linguini1, 2025-04-27:
> @linguini1 as you are most familiar with rPI-Pico can we assign you this task to research in a free moment? :-) Having this single usb port flashing just as with esptool would enable automated testing on a real world hardware :-)

I can certainly try to research how this is done!

cederom, 2025-04-27:
TANK U SIR! :-)

linguini1, 2025-05-05:
After doing more research on this topic I haven't been able to find any mention of automatic flashing of RP2040 chips. All sources I've found said that the chip requires the "bootsel" button to be held to enter boot mode, or people have created custom boot loaders to flash without holding the button.

It was also suggested on this forum to use the SWD pins: https://forums.raspberrypi.com/viewtopic.php?t=328795

The pico SDK is meant to have some utilities for rebooting into `bootsel`, so it may be possible for the purposes of mass testing to create some bootloader/helper program that reboots the pico into bootsel, uploads the configuration to be tested, tests it, and reboots into bootsel again to repeat.

keever50, 2025-05-05:
I can confirm platformio (https://github.com/earlephilhower/arduino-pico) is including a bootloader.
Might be a nice addition for NuttX to add it as well.

cederom, 2025-05-05:
> @keever50: I can confirm platformio (https://github.com/earlephilhower/arduino-pico) is including a bootloader. Might be a nice addition for NuttX to add it as well.

Hmm this seems to be whole Arduino SDK.. we would prefer to have our own bootloader and stay external SDK independent right? :-P

linguini1, 2025-05-09:
If we were to write such a bootloader for NuttX, would it be preferable to:

1) Have a bootloader that puts the RP2040 into BOOTSEL when it detects something like 1200 baud on a UART line
2) Have an application/NSH command that when run reboots the RP2040 into bootsel
3) Expect some kind of special bytes over USB to enter BOOTSEL 

My thought is that with 2) we could flash the Pico manually once, and then always be able to reboot into bootsel to flash more configurations, which would be suitable for creating test-farm setups. I think 1) will be inconvenient because it requires a UART connection (as opposed to just a USB connection which is used for both powering and interacting with the RP2040 in a lot of these boards) and I think 3) would be complicated to get right/integrate + require some special byte sequence known in advance and sent by the host device.

keever50, 2025-05-10:
Just a thought but you can use a raspberry pi debug probe (or another pico) to program the pico over SWD.
The probe firmware is open source so you can tweak it to give it an unique identifier. https://github.com/raspberrypi/debugprobe
(I think its also possible to get an unique ID of the RP2040 to be flashed over SWD?)

Or create a microcontroller dedicated to reset the pico into bootloader and push firmware onto it via usb,
or via SWD. (or mod the debugprobe to do this)

Those are dumb but easier solutions. We dont want to make it too complex of course. Maybe worth a thought.

cederom, 2025-05-10:
Yeah but the goal (and need) is to have rpi using only single usb connection to flash and run.. to be honest its quite funny it cannot by default :D

shtirlic, 2025-08-14:
@cederom  I must clarify that this is NOT `usb-protocol-reset-vector`  it's calling internal pico bootrom functions to reboot normally or reboot to bootsel mode, so you can just type/send the `reboot bootloader` command to nsh and it goes to reboot. The only problem I see if the NuttX crashed or not booting at all so the board will be stuck.

cederom, 2025-08-16:
Do we have `make flash` target implemented @shtirlic ? :-)

Today I have some free moment will test the reset vector and flashing :-) Thanks again :-)

cederom, 2025-08-19:
Not yet implemented. `make flash` only produces `nuttx.uf2` while the goal is to automate flashing of the physical chip like in other targets. Reopening :-)

linguini1, 2025-09-07:
> > @linguini1 Have a bootloader that puts the RP2040 into BOOTSEL when it detects something like 1200 baud on a UART line
> 
> In my RP2040 board I have implemented this over USB CDC-ACM instead of UART. This comes at the cost of having a background demon that listens for LINE_CODING events (let me improperly call them "baud rate changes"), though.

> This way I can run `stty -F /dev/serial/by-id/usb-<your_device_id> 1200` on the host and have the board reboot to bootloader, and then it's just a matter of a few shell commands to wait for the USB drive to appear and copy the new UF2 file to it. I've wrapped all these actions in a couple of [Just](https://just.systems/man/en/) recipes for ease of use, so I can get away with things like `just build && just bootsel-flash` :)

I really like this approach! If it can be Kconfig selected as whether to run at boot or not for the RP2040s, I think it's great. That way the Pico can have a single connection (USB shell) for testing and booting in a hardware test setup. Thanks for sharing that!