apache/nuttx #16278: Actions list to improve NuttX quality and reliability

acassis, 2025-04-28:
### Is your feature request related to a problem? Please describe.

These are some of suggested actions we need to take to improve NuttX quality:

1. **Automated Test and CI Integration** (only will cover some corner cases)
2. **Test Coverage Metrics**
3. **Expand and Improve Documentation**
4. **Standardize Board Port Quality**
5. **Better Unit Testing and Mocking**
6. **Stable API Guarantees**
7. **Regression Testing**
8. **Perform Benchmarks Metrics**
9. **Create Templates to use as reference to Documentation**
10. **Static Analysis into CI**
11. Create a base board with standard MCU plug/insert connector to test peripherals in more microcontrollers

### Describe the solution you'd like

This is a summary of actions we need to take to improve NuttX quality and reliability

### Describe alternatives you've considered

No alternatives

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-04-28:
1. **Automated Test and CI Integration**

It will help to detect for example if the board is not starting and if some testings (ostest, etc) is passing.
There are many initiatives to have a Automated Test integrated with CI:

- Lup Yuen Test Bot and CI: https://lupyuen.org/articles/testbot2.html and https://lupyuen.codeberg.page/articles/ci.html
- Tomek NXDART (only proposal until now)
- Espressif NuttX Test Farm using Robot Framework
- Xiaomi Internal Test Farm (no info)

acassis, 2025-04-28:
2. **Test Coverage Metrics**

Integrate code coverage tools like gcov/lcov for unit tests, dhrystone, coremark, etc
Display and track code coverage over time to identify untested parts of the kernel, drivers, and libraries.

acassis, 2025-04-28:
4. **Standardize Board Port Quality**

Create a checklist for each board port to ensure minimum quality:
        Does ostest pass?
        Do basic drivers (UART, Timer, GPIO) work?
        Is SMP tested (if applicable)?
Boards that don't meet the minimum criteria are marked as "experimental" or "unsupported".

acassis, 2025-04-28:
5. **Better Unit Testing and Mocking**

Expand the apps/testing suite with more unit tests.
Use frameworks like CMocka or extend the existing ostest usage.
Mock drivers and hardware to allow kernel logic testing even without hardware.

acassis, 2025-04-28:
6. **Stable API Guarantees**

Formalize API stability between releases (similar to "Stable API" policy in Linux kernel).
Document which APIs are considered stable and which are still experimental.
Add a deprecation process for removing/renaming public APIs.

acassis, 2025-04-28:
7. **Regression Testing**

Maintain a regression test suite to ensure that previously fixed bugs do not come back.
Basically when someone found an issue they should create a test to be integrated into ostest to detect it in the future.
Set up automatic re-run of regression tests in CI when code is merged.

acassis, 2025-04-28:
8. **Perform Benchmarks Metrics**

Create standard performance tests:
Boot time benchmarks
Context switch time
Interrupt latency
Track performance regressions automatically in CI.

acassis, 2025-04-28:
9. **Create Templates for Documentation**

The idea is to have standard reference documents to be used when adding documentatio for a new Boards, a new Driver, etc

acassis, 2025-04-28:
10. **Static Analysis into CI**

Using static analysis we can find simple mistakes that might have been introduced.
Things like a function that forgot to return a value in some code path, or
use of uninitialized variable, can be caught by static analysis.

acassis, 2025-04-28:
11. **Create a base board with a standard MCU plug/insert connector**

The idea is to have a single base board with many peripherals (sensors, display, audio codec, ethernet, etc) and that board should use some standard MCU plug/insert board connector (like RPi CM4S or SparkFun MicroMod).

This way we can test existent peripheral drivers in many different microcontrollers. Also that base board (mainboard) could have features to facilitates HW tests.

Jasinsky, 2025-05-06:
> 11. **Create a base board with a standard MCU plug/insert connector**
>
> The idea is to have a single base board with many peripherals (sensors, display, audio codec, ethernet, etc) and that board [...]

Hello, @acassis - 
just want to mention that products like that already exist - 
https://www.mikroe.com/fusion-for-arm#main-buy-field
https://www.mikroe.com/development-boards/arm-32-bit

a bit pricey, but  can save some hardware development time