apache/nuttx #16290: [BUG] ostest failure on Nucleo-G071RB NSH config

stbenn, 2025-04-29:
### Description / Steps to reproduce the issue

On the Nucleo-G071RB nsh configuration, running the ostest fails.

Steps to reproduce:

1. `make distclean -j`
2. `./tools/configure.sh -l nucleo-g071rb:nsh`
3. `make menuconfig`
4. Enable the `ostest` application
5. Exit and save
6. Compile and program board.
7. In nsh, run ostest

Log file of console interaction:
[devttyACM1_2025_04_29.10.43.51.422.txt](https://github.com/user-attachments/files/19962449/devttyACM1_2025_04_29.10.43.51.422.txt)

I also tried with debug symbols enabled, and disabling full optimization. The failure still occurred. 

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.04 docker container on Ubuntu 24.04 host

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Board support]

### Host information

file sysinfo.h not exists
NuttX CFLAGS:
  -Wstrict-prototypes
  -Wno-attributes
  -Wno-unknown-pragmas
  -Wno-psabi
  -Os
  -fno-strict-aliasing
  -fomit-frame-pointer
  --param=min-pagesize=0
  -fno-common
  -Wall
  -Wshadow
  -Wundef
  -ffunction-sections
  -fdata-sections
  -mlittle-endian 
  -mcpu=cortex-m0
  -mthumb
  -mfloat-abi=soft
  -mthumb
  -Wa-mthumb
  -Wa-mimplicit-it=always
  -isystem /home/tbennett/nuttxspace/nuttx/include
  -D__NuttX__
  -DNDEBUG 

NuttX CXXFLAGS:
  -Wno-attributes
[... 282 lines cut ...]
  CONFIG_TESTING_OSTEST=y
  CONFIG_TESTING_OSTEST_LOOPS=1
  CONFIG_TESTING_OSTEST_STACKSIZE=8192
  CONFIG_TESTING_OSTEST_NBARRIER_THREADS=8
  CONFIG_TESTING_OSTEST_RR_RANGE=10000
  CONFIG_TESTING_OSTEST_RR_RUNS=10
  CONFIG_TESTING_OSTEST_WAITRESULT=y

Host system OS:
  Linux 77e556bcb78b 5.15.167.4-microsoft-standard-WSL2 #1 SMP Tue Nov 5 00:21:55 UTC 2024 x86_64 x86_64

Host system PATH:
  /home/ubuntu/.vscode-server/bin/17baf841131aa23349f217ca7c570c76ee87b957/bin/remote-cli
  /usr/bin/arm-none-eabi/bin
  /usr/local/bin
  /usr/local/sbin
  /usr/local/bin
  /usr/sbin
  /usr/bin
  /sbin
  /bin

### Verification

- [x] I have verified before submitting the report.

stbenn, 2025-04-29:
@raiden00pl Thank you for looking into this! I wanted to verify this behavior as I am currently working on adding board support for the Nucleo G0B1RE, and intended to use `ostest` as one of the criteria for acceptance due to the recently reported issue by Alan.

anjiahao1, 2025-04-30:
https://nuttx.apache.org/docs/latest/debugging/gdbserver.html
@stbenn try use this debug tools

closed 2025-05-06 by stbenn