apache/nuttx #16341: [BUG] System crashes after https://github.com/apache/nuttx/pull/16231

jlaitine, 2025-05-08:
### Description / Steps to reproduce the issue

To reproduce:

./tools/configure.sh sim:citest
make
./nuttx

nsh> cmocka

<clip>

```

Output:
[==========] nuttx_mm_test_suites: 8 test(s) run.
[  PASSED  ] 8 test(s).
[==========] tests: Running 1 test(s).
[ RUN      ] drivertest_posix_timer
[  ERROR   ] --- 2019 is not within the range [1990, 2010]
[   LINE   ] --- drivertest_posix_timer.c:162: error: Failure!
[  FAILED  ] drivertest_posix_timer
[==========] tests: 1 test(s) run.
[  PASSED  ] 0 test(s).
[  FAILED  ] tests: 1 test(s), listed below:
[  FAILED  ] drivertest_posix_timer

 1 FAILED TEST(S)
Segmentation fault (core dumped)

```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 22.04.1

### NuttX Version

master

### Issue Architecture

[Arch: simulator]

### Issue Area

[Area: Other]

### Host information

NuttX CFLAGS:
  --g\
  -fno-omit-frame-pointer
  -fno-optimize-sibling-calls
  -fno-common
  -fvisibility=hidden
  -ffunction-sections
  -fdata-sections
  -Wall
  -Wstrict-prototypes
  -Wshadow
  -Wundef
  -Wno-attributes
  -Wno-unknown-pragmas
  -fno-pic
  -mcmodel=medium
[... 580 lines cut ...]
  CONFIG_CM_PTHREAD_TEST=y

Host system OS:
  Ubuntu 22.04.4 LTS Linux jlaitine-thinkpad 6.8.0-59-generic #61~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 15 17:03:15 UTC 2 x86_64 x86_64

Host system PATH:
  /home/jlaitine/.cargo/bin
  /opt/gcc-arm-none-eabi-9-2020-q2-update/bin
  /home/jlaitine/.local/bin
  /home/jlaitine/bin
  /opt/gradle/latest/bin
  /usr/local/sbin
  /usr/local/bin
  /usr/sbin
  /usr/bin
  /sbin
  /bin
  /usr/games
  /usr/local/games
  /snap/bin
  /home/jlaitine/bin

### Verification

- [x] I have verified before submitting the report.

jlaitine, 2025-05-08:
Specifically, I bisected the offending patch and it is 900b1c19ddfaece35df7306ad368db72c6a32a78

On the next patch, 9dbb9b49c65ed738948f3047d44d871c195cf3d2, the sim doesn't even start any more:

 ./nuttx 
CHelloWorld: Constructor: mSecret=42
nsh: init: open failed: 2
Segmentation fault (core dumped)

Fix-Point, 2025-05-08:
Hi. The issue is caused by the `work->qtime += 1` in `libs/libc/wqueue/work_queue.c`. The user workqueue using `nxsem_timedwait`, which has already taken the insufficient ticks into consideration.  After removing `work->qtime += 1`, PR https://github.com/apache/nuttx/pull/16342 should have fixed this issue.

jlaitine, 2025-05-08:
Unfortunately, it looks like 16342 won't fix the issue;

I cherry-picked the following patches:

```

commit 878c45eba94d96997a3e737696167b99452c0f37 (HEAD)
Author: ouyangxiangzhen <ouyangxiangzhen@xiaomi.com>
Date:   Thu May 8 11:24:46 2025 +0800

    sched/wqueue: Improve performance of the work_queue.
    
    This commit improve the performance of the work_queue by reducing
    unnecessary wdog timer setting.
    
    Signed-off-by: ouyangxiangzhen <ouyangxiangzhen@xiaomi.com>

commit 47c3c1640fdc7dfe084a53e68c12f4b3557fa87a
Author: ouyangxiangzhen <ouyangxiangzhen@xiaomi.com>
Date:   Thu May 8 11:17:46 2025 +0800

    sched/wqueue: Fix work_cancel_sync.
    
    This commit fixed work_cancel_sync at a very rare boundary case. When a worker thread re-enqueues the work data structure during the execution of work, the user thread cannot directly dequeue the work in work_cancel_sync. Instead, it should wait until all workers' references to the work data structure have been eliminated after dequeuing.
    
    Signed-off-by: ouyangxiangzhen <ouyangxiangzhen@xiaomi.com>

```

On top of current master (156469f158753db41d39d08008aa1ae978af1a47)

And the result is the same:

```

[==========] nuttx_mm_test_suites: 8 test(s) run.
[  PASSED  ] 8 test(s).
[==========] tests: Running 1 test(s).
[ RUN      ] drivertest_posix_timer
[  ERROR   ] --- 2015 is not within the range [1990, 2010]
[   LINE   ] --- drivertest_posix_timer.c:162: error: Failure!
[  FAILED  ] drivertest_posix_timer
[==========] tests: 1 test(s) run.
[  PASSED  ] 0 test(s).
[  FAILED  ] tests: 1 test(s), listed below:
[  FAILED  ] drivertest_posix_timer

 1 FAILED TEST(S)
Segmentation fault (core dumped)

```

Fix-Point, 2025-05-08:
Interesting, I checkout the last commit (c22df41ca6a17bcfc10c242242b0cd55945db8ab) before the https://github.com/apache/nuttx/pull/16231. It failed the same test cases ocassionally.

```
The current real time: sec is 946684884, nsec is 959353016
sleep 2 seconds
Setting time: sec is 946684886, nsec is 959353016
get real time clock again
Obtaining the current time after setting: sec = 946684886, nsec = 959403366
[       OK ] test_nuttx_clock_test_clock01
[ RUN      ] test_nuttx_clock_test_clock02
[       OK ] test_nuttx_clock_test_clock02
[==========] nuttx_time_test_suites: 7 test(s) run.
[  PASSED  ] 6 test(s).
[  FAILED  ] nuttx_time_test_suites: 1 test(s), listed below:
[  FAILED  ] test_nuttx_clock_test_timer03

 1 FAILED TEST(S)
Cmocka Test Completed.
nsh> uname -a
NuttX  12.7.2-vela c22df41ca6a May  8 2025 19:43:28 sim sim
```
```

```
[  PASSED  ] 8 test(s).
[==========] tests: Running 1 test(s).
[ RUN      ] drivertest_posix_timer
[  ERROR   ] --- 2020 is not within the range [1990, 2010]
[   LINE   ] --- drivertest_posix_timer.c:162: error: Failure!
[  FAILED  ] drivertest_posix_timer
[==========] tests: 1 test(s) run.
[  PASSED  ] 0 test(s).
[  FAILED  ] tests: 1 test(s), listed below:
[  FAILED  ] drivertest_posix_timer

 1 FAILED TEST(S)
[1]    1618803 segmentation fault (core dumped)  ./nuttx
```

Fix-Point, 2025-05-09:
> Interesting, then it is probably some older bug, which just appears when timings change!

Can you check if this PR https://github.com/apache/nuttx/pull/16343 fixed the bug?

jlaitine, 2025-05-12:
btw. @anchao this was closed wrongly, PR's #16342 or #16343 won't fix this. #16358 will.

xiaoxiang781216, 2025-05-12:
let's reopen it

anchao, 2025-05-12:
> btw. [@anchao](https://github.com/anchao) this was closed wrongly, PR's [#16342](https://github.com/apache/nuttx/pull/16342) or [#16343](https://github.com/apache/nuttx/pull/16343) won't fix this. [#16358](https://github.com/apache/nuttx/pull/16358) will.

Sorry, #16342 solved my issue but I didn’t notice that this PR was linked to this issue.

jlaitine, 2025-05-23:
This is fixed by PR #16358 and can now be closed

closed 2025-05-23 by jlaitine