apache/nuttx #16345: [BUG] USBDEV CDC-NCM only works on first connect

TimJTi, 2025-05-08:
### Description / Steps to reproduce the issue

I have found that the CDC-NCM driver only works the first time my board's USB cable is connection to the host PC. If the USB is unplugged then replugged the host reports the connection as Disconnected.

I have found:

- This is true either for CDC-NCM as the only USB device driver, or if configured as a composite device.
- If configured as a composite device, with MSD as the other composite driver, the MSD works correctly with USB unplug/re-plug cycles
- Look at the MSD code, there are calls to disconnect functions that is also called at in cdc-ncm, but there is no meaningful code there
    - As an aside, the reset is actually called on a connect not a disconnect, both for MSD and CDC-NCM!!??
- I hacked in some disconnect code loosely based on what the MSD driver does and called `cdcncm_resetconfig` but found:
    - The calls to EP_DISABLE caused the code to crash
    - commenting those out wasn't a magic fix as I suspect the configuration needs to be set up again with some call from somewhere

I have admitted defeat as USB and CDC-NCM are out of my league.

Can anyone:
a) Confirm this behaviour on another board/arch as I only have access to my custom board which is a bit odd when it comes to USB
b) and/or suggest a possible fix - happy to try things as always. Maybe a suggestion of what might need to go in that reset function?

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

N/A

### NuttX Version

Master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Drivers]

### Verification

- [x] I have verified before submitting the report.

TimJTi, 2025-05-12:
> Hey if things behave a bit weird, ensure that the stacksizes are configured big enough. This is something you can easily confirm by enabling stack canaries and running 'ps' in nsh. If a thread is around or over 80%, it's probably a good idea to make it bigger.
> 
> I had problems like this before and this solved it. Even in default configurations.

I'll give it a go - thanks for the suggestion :-)

TimJTi, 2025-05-13:
@keever50 sadly, not to do with stack size or anything. I am pretty sure the driver is missing some deinit/unbind type code.

Hopefully someone who has worked with CDC-NCM will be along soon  :-)

xiaoxiang781216, 2025-05-13:
@zhhyu7 could you look at this problem?

zhhyu7, 2025-05-14:
> [@xiaoxiang781216](https://github.com/xiaoxiang781216)
> 
> > [@zhhyu7](https://github.com/zhhyu7) could you look at this problem?
> 
> Thanks! Let me know if you want any trace info etc., or for me to try anything - I have been trying to work through this by comparison to CDC-ECM (which works) but haven't found anything.

@TimJTi Could you help try to only reset the config in the cdcncm_disconnect method, something like the following:

```
static void cdcncm_disconnect(FAR struct usbdevclass_driver_s *driver,    
                              FAR struct usbdev_s *dev)                   
{                                                                         
  FAR struct cdcncm_driver_s *self = (FAR struct cdcncm_driver_s *)driver;
  self->config = CDCECM_CONFIGID_NONE;                                    
  uinfo("\n");                                                            
}                                                                         
```

Is there a complete backtrace for the previously mentioned "EP_DISABLE caused the code to crash"? Is it possible that the "ep_disable callback" in the usb driver access the released resources?

TimJTi, 2025-05-14:
> [@TimJTi](https://github.com/TimJTi) Could you help try to only reset the config in the cdcncm_disconnect method, something like the following:
>
> ```
> static void cdcncm_disconnect(FAR struct [...]

@xiaoxiang781216 @zhhyu7 

That works!

Since there is also no code in the CDC-ECM disconnect function, I'd tried hacking in the code from the CDC-ECM `reset_config` function:
```

  /* Are we configured? */

  if (self->config != CDCECM_CONFIGID_NONE)
    {
      /* Yes.. but not anymore */

      self->config = CDCECM_CONFIGID_NONE;

      /* Inform the networking layer that the link is down */

      self->dev.d_ifdown(&self->dev);

      /* Disable endpoints.  This should force completion of all pending
       * transfers.
       */

      EP_DISABLE(self->epint);
      EP_DISABLE(self->epbulkin);
      EP_DISABLE(self->epbulkout);
    }
}
```
The guard perhaps meant it ran when it shouldn't and vice versa?

the EP_DISABLE calls don't crash the board either if I add those to your code for the `cdcncm_disconnect` function - should we have them here?

I am happy to submit a PR for this once you can confirm the preferred function steps: at present I have:

```
static void cdcncm_disconnect(FAR struct usbdevclass_driver_s *driver,
                              FAR struct usbdev_s *dev)
{
  FAR struct cdcncm_driver_s *self = (FAR struct cdcncm_driver_s *)driver;
  self->config = CDCECM_CONFIGID_NONE;   

  EP_DISABLE(self->epint);
  EP_DISABLE(self->epbulkin);
  EP_DISABLE(self->epbulkout);

  uinfo("cdcncm disconnect\n");
}
```

zhhyu7, 2025-05-14:
@TimJTi Would the crash mentioned earlier occur in code like the following:
```
static void cdcncm_disconnect(FAR struct usbdevclass_driver_s *driver,    
                              FAR struct usbdev_s *dev)                   
{                                                                         
  FAR struct cdcncm_driver_s *self = (FAR struct cdcncm_driver_s *)driver;
  cdcncm_resetconfig(self);                                    
  uinfo("\n");                                                            
}     
```
If not, then let's add ```cdcncm_resetconfig``` in ```cdcncm_disconnect```.

TimJTi, 2025-05-14:
@zhhyu7 @xiaoxiang781216

That works fine - who knows what I did when I hacked a quick not-fix lol.

Thank you very much for taking the time to assist 👍 

I will be using CDC-NCM with lots of unplug/replug in the next few days, so if it remains stable I will submit the PR.

I also see a compiler warning cdcncm is NOT used as a composite device - just some #ifdefs not in the right place. I will try and fix that at the same time

TimJTi, 2025-05-14:
> I also see a compiler warning cdcncm is NOT used as a composite device - just some #ifdefs not in the right place. I will try and fix that at the same time

It seems there is some confusion over what function are needed public what are private and what *might* be called in non-composite setups. It is equally true of CDC-NCM and probably dates back to basing code on CDC-ACM!

It is not right to include any work on that in a PR for this issue...but it might fail CI in which case it will have to be tackled!

TimJTi, 2025-05-23:
Have now submitted the [PR](https://github.com/apache/nuttx/pull/16435/checks) for this 👍

closed 2025-05-24 by xiaoxiang781216