apache/nuttx #16357: [esp32] <wifi set softap, can't work>

461911662, 2025-05-11:
### Description / Steps to reproduce the issue

wifi set softap, can't work.
**my config:** 
./tools/configure.sh -a vendor/boss/app/nuttx-apps esp32-devkitc:wifi && make -j4
flash nuttx
**follow the log:**
Hard resetting via RTS pin...
--- Using autodetected port /dev/ttyUSB0
--- serial_com 1.5.0 on /dev/ttyUSB0 115200
1150 mmu set 00020000, pos 00020000
entry 0x400859ec
*** Booting NuttX ***
I (78) boot: chip revision: v3.0
I (79) boot.esp32: SPI Speed      : 40MHz
I (79) boot.esp32: SPI Mode       : DIO
I (80) boot.esp32: SPI Flash Size : 16MB
I (84) boot: Enabling RNG early entropy source...
dram: lma 0x00001020 vma 0x3ffbe7d0 len 0x36e4   (14052)
iram: lma 0x0000470c vma 0x40080000 len 0x1fe70  (130672)
padd: lma 0x00024588 vma 0x00000000 len 0xba70   (47728)
imap: lma 0x00030000 vma 0x400f0000 len 0x6a9e8  (436712)
padd: lma 0x0009a9f0 vma 0x00000000 len 0x5628   (22056)
dmap: lma 0x000a0020 vma 0x3f400020 len 0x16048  (90184)
total segments stored 6
A__esp32_start: ESP32 chip revision is v3.0
BI (77) wifi:wifi firmware version: 34922fa
I (77) wifi:wifi certification version: v7.0
I (77) wifi:config NVS flash: disabled
I (78) wifi:config nano formating: disabled
I (78) wifi:Init data frame dynamic rx buffer num: 32
I (78) wifi:Init static rx mgmt buffer num: 5
I (78) wifi:Init management short buffer numesp_wifi_adapter_init: OK to initialize Wi-Fi adapter
esp32_wlan_softap_initialize: Wi-Fi softAP MAC: 24:0A:C4:E8:03:9D
netdev_register: Registered MAC: 24:0a:c4:e8:03:9d as dev: wlan0
esp32_net_initialize: INFO: Initialize Wi-Fi adapter No.0 success
esp32_wlan_softap_initialize: INFO: Initialize Wi-Fi softAP net success
netdev_ifr_ioctl: cmd: 1794
netdev_ifr_ioctl: cmd: 1796
netdev_ifr_ioctl: cmd: 1800
netdev_ifr_ioctl: cmd: 1818
wlan_ifup: Bringing up: 10.0.0.2
esp_phy_get_init_data: Loading PHY init data from application binary
: 32
I (83) wifi:Init dynamic tx buffer num: 32
I (83) wifi:Init static rx buffer size: 1600
I (83) wifi:Init static rx buffer num: 10
I (94) wifi:Init dynamic rx buffer num: 32
I (136) phy_init: phy_version 4830,54550f7,Jun 20 2024,14:22:08
wifi_set_intr: cpu_no=0, intr_source=0, intr_num=0, intr_prio=1
esp_set_isr: n=0 f=0x40085770 arg=0 irq=5
esp32_ints_on: INFO mask=00000001 irq=0
I (193) wifi:modeesp_event_post: Event: base=WIFI_EVENT id=12 data=0 data_size=0 ticks=4294967295
esp_wifi_softap_start: OK to start Wi-Fi SoftAP
nsh_telnetstart: Starting the Telnet daemon
esp_evt_work_cb: INFO: Wi-Fi softap start
telnetd_daemon: Accepting connections on port 23
 : softAP (24:0a:c4:e8:03:9d)
I (193) wifi:Total power save buffer number: 16
esp_evt_work_cb: INFO: Set ps type=0netd [5:100]

NuttShell (NSH)
nsh> eacon: 752/752
I (195) wifi:Init max length of beacon: 752/752
I (218) wifi:Set ps type: 0, coexist: 0

I (17136) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, pesp_event_post: Event: base=WIFI_EVENT id=14 data=0x3ffc7450 data_size=8 ticks=4294967295
esp_evt_work_cb: INFO: Wi-Fi station join
rof:1
I (17137) wifi:station: 62:62:c6:ee:90:60 join, AID=1, bgn, 40U
[... 224 lines cut ...]
I (895

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Linux DESKTOP-BNU9SN9 5.15.167.4-microsoft-standard-WSL2 #1 SMP Tue Nov 5 00:21:55 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Drivers]

### Verification

- [x] I have verified before submitting the report.

461911662, 2025-05-11:
**Current defconfig:**
liangliang@DESKTOP-BNU9SN9:~/wk/boss/nuttx_va$ cat defconfig
#
# This file is autogenerated: PLEASE DO NOT EDIT IT.
#
# You can use "make menuconfig" to make any modifications to the installed .config file.
# You can then do "make savedefconfig" to generate a new defconfig file that includes your
# modifications.
#
# CONFIG_ARCH_LEDS is not set
# CONFIG_NDEBUG is not set
# CONFIG_NSH_ARGCAT is not set
# CONFIG_NSH_CMDOPT_HEXDUMP is not set
CONFIG_ALLOW_BSD_COMPONENTS=y
CONFIG_ARCH="xtensa"
CONFIG_ARCH_BOARD="esp32-devkitc"
CONFIG_ARCH_BOARD_COMMON=y
CONFIG_ARCH_BOARD_ESP32_DEVKITC=y
CONFIG_ARCH_CHIP="esp32"
CONFIG_ARCH_CHIP_ESP32=y
CONFIG_ARCH_CHIP_ESP32WROVER=y
CONFIG_ARCH_INTERRUPTSTACK=2048
CONFIG_ARCH_STACKDUMP=y
CONFIG_ARCH_XTENSA=y
CONFIG_BOARD_LOOPSPERMSEC=16717
CONFIG_BUILTIN=y
CONFIG_DEBUG_FEATURES=y
CONFIG_DEBUG_NET=y
CONFIG_DEBUG_NET_ERROR=y
CONFIG_DEBUG_NET_INFO=y
CONFIG_DEBUG_NET_WARN=y
CONFIG_DEBUG_WIRELESS=y
CONFIG_DEBUG_WIRELESS_ERROR=y
CONFIG_DEBUG_WIRELESS_INFO=y
CONFIG_DEBUG_WIRELESS_WARN=y
CONFIG_DEFAULT_TASK_STACKSIZE=4096
CONFIG_DEV_URANDOM=y
CONFIG_DRIVERS_IEEE80211=y
CONFIG_DRIVERS_WIRELESS=y
CONFIG_ESP32_SPIFLASH=y
CONFIG_ESP32_SPIFLASH_SPIFFS=y
CONFIG_ESP32_STORAGE_MTD_SIZE=0x80000
CONFIG_ESP32_UART0=y
CONFIG_ESP32_WIFI=y
CONFIG_ESP32_WIFI_SOFTAP=y
CONFIG_ESPRESSIF_3RDPARTY_OFFLINE=y
CONFIG_ESPRESSIF_3RDPARTY_TOOLS_PATH="vendor/boss/xtensa/esp-hal-3rdparty"
CONFIG_FS_PROCFS=y
CONFIG_IDLETHREAD_STACKSIZE=3072
CONFIG_INIT_ENTRYPOINT="nsh_main"
CONFIG_INIT_STACKSIZE=3072
CONFIG_INTELHEX_BINARY=y
CONFIG_MM_REGIONS=3
CONFIG_NAME_MAX=48
CONFIG_NETDB_DNSCLIENT=y
CONFIG_NETDB_DNSCLIENT_NAMESIZE=64
CONFIG_NETDEV_LATEINIT=y
CONFIG_NETDEV_PHY_IOCTL=y
CONFIG_NETDEV_WIRELESS_IOCTL=y
CONFIG_NETUTILS_CJSON=y
CONFIG_NETUTILS_IPERF=y
CONFIG_NETUTILS_TELNETD=y
CONFIG_NET_BROADCAST=y
CONFIG_NET_ETH_PKTSIZE=1518
CONFIG_NET_ICMP_SOCKET=y
CONFIG_NET_PREALLOC_DEVIF_CALLBACKS=32
CONFIG_NET_STATISTICS=y
CONFIG_NET_TCP=y
CONFIG_NET_TCP_DELAYED_ACK=y
CONFIG_NET_TCP_WRITE_BUFFERS=y
CONFIG_NET_UDP=y
CONFIG_NSH_ARCHINIT=y
CONFIG_NSH_BUILTIN_APPS=y
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_LINELEN=300
CONFIG_NSH_READLINE=y
CONFIG_POSIX_SPAWN_DEFAULT_STACKSIZE=2048
CONFIG_PREALLOC_TIMERS=4
CONFIG_PTHREAD_MUTEX_TYPES=y
CONFIG_RAM_SIZE=114688
CONFIG_RAM_START=0x20000000
CONFIG_RR_INTERVAL=200
CONFIG_SCHED_LPWORK=y
CONFIG_SCHED_WAITPID=y
CONFIG_SIG_DEFAULT=y
CONFIG_SPI=y
CONFIG_SPIFFS_NAME_MAX=48
CONFIG_START_DAY=6
CONFIG_START_MONTH=12
CONFIG_START_YEAR=2011
CONFIG_SYSLOG_BUFFER=y
CONFIG_SYSTEM_DHCPC_RENEW=y
CONFIG_SYSTEM_NSH=y
CONFIG_SYSTEM_PING=y
CONFIG_TLS_TASK_NELEM=4
CONFIG_UART0_SERIAL_CONSOLE=y
CONFIG_WIRELESS=y
CONFIG_WIRELESS_WAPI=y
CONFIG_WIRELESS_WAPI_CMDTOOL=y
CONFIG_WIRELESS_WAPI_INITCONF=y
liangliang@DESKTOP-BNU9SN9:~/wk/boss/nuttx_va$

eren-terzioglu, 2025-05-12:
Hi,

Please update your master branch, after https://github.com/apache/nuttx/pull/16342 it should work.

461911662, 2025-05-12:
这是来自QQ邮箱的假期自动回复邮件。你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。

closed 2025-05-15 by xiaoxiang781216