apache/nuttx #16367: [BUG] NXStyle is not catching brackets starting in the same line

acassis, 2025-05-13:
### Description / Steps to reproduce the issue

[image]

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

N/A

### NuttX Version

mainline

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Other]

### Verification

- [x] I have verified before submitting the report.

TimJTi, 2025-05-13:
Locally? CI? Steps to reproduce?

Works for me using checkpatch.sh

acassis, 2025-05-13:
Hi @TimJTi this print is from AVR PR: https://github.com/apache/nuttx/pull/16365

acassis, 2025-07-14:
@TimJTi good question! I think it used to work in the past, because Greg implemented nxstyle to be consistent with the Coding Style guy, but we need to confirm it, for example downloading an old nuttx release and testing nxstyle against this same file.

closed 2026-10-15 by xiaoxiang781216