apache/nuttx #16373: [HELP] SPI TX DMA complete callback

dakejahl, 2025-05-13:
### Description

I don't see a way to install a callback from user space for SPI DMA TX complete. Is this possible to do?

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-05-17:
Hi @dakejahl it is not directly possible to install a callback from user space for SPI DMA completion due the userspace/kernel space separation (in FLAT mode it could work, but it will neither work in PROTECT mode or KERNEL mode).

Maybe you can modify your spi driver to create a notification event when DMA TX is complete in your chip and use some mechanism such as signal, poll, message queue, etc to notify your application. Other option is create a character driver that subscribe to DMA TX complete IRQ and do this notification, this way we avoid modifying the existing driver, but there are some complications since the IRQ is already handled in the spi driver.

If you want to use signal, you can see the driver nuttx/drivers/sensors/zerocross.c when I used it and the apps/examples/zerocross application example.

dakejahl, 2025-05-20:
Thanks for the reply! Okay yeah that makes sense. Using signals makes sense, there's a sem already in use for the STM32H7 SPI RX complete. Unfortunately I'm dealing with a pre-compiled library + API that imposes tight timing requirements, a supplied callback must be invoked <60us after the last SPI TX frame (before the DRDY fires and calls another callback, and the order of callbacks matters). I temporarily solved the issue by increasing the thread priority of my driver such that it's not pre-empted between SPI_EXCHANGE and callback() and instead benefits from pre-emption once the SPI_EXCHANGE is complete. It's not ideal since the driver shouldn't actually be high priority. I'll have to think about this harder.