apache/nuttx #16386: [BUG] Potential pointer arithmetic issue?

emomaxd, 2025-05-14:
### Description / Steps to reproduce the issue

When I do anything with g_idle_topstack—casting, arithmetic, assignment etc.—its value suddenly becomes 1.

In the first image, you can see the issue. I added extra lines for debugging purposes, but the result stays the same. The stk variable holds the value after casting.

In the second image, I included where g_idle_topstack is defined.

I resolved the issue by taking the address of g_idle_topstack using the & operator.

Here's the original code (in case it helps):
```shell
tcb->stack_alloc_ptr = (void *)(g_idle_topstack -
                                      CONFIG_IDLETHREAD_STACKSIZE);
      tcb->stack_base_ptr  = tcb->stack_alloc_ptr;
      tcb->adj_stack_size  = CONFIG_IDLETHREAD_STACKSIZE;
```

--- 

And here's the fix that worked:
```shell
tcb->stack_alloc_ptr = (void *)(&g_idle_topstack -
                                      CONFIG_IDLETHREAD_STACKSIZE);
      tcb->stack_base_ptr  = tcb->stack_alloc_ptr;
      tcb->adj_stack_size  = CONFIG_IDLETHREAD_STACKSIZE;
```

Can u help with this?

<table>
  <tr>
    <td>[image]</td>
    <td>[image]</td>
  </tr>
  <tr>
    <td align="center"><b>Debug View</b></td>
    <td align="center"><b>Definition of <code>g_idle_topstack</code></b></td>
  </tr>
</table>

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.04

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Kernel]

### Verification

- [x] I have verified before submitting the report.

pkarashchenko, 2025-05-14:
Could you please specify the board example that can be used to reproduce this issue?
I see that for `arm` the `g_idle_topstack` is defined as `EXTERN const uintptr_t g_idle_topstack;`, so basically the `g_idle_topstack` symbol points to a memory cell that contains the address value of the top of the stack. The math in `tcb->stack_alloc_ptr = (void *)(g_idle_topstack - CONFIG_IDLETHREAD_STACKSIZE);` is correct since the numeric value is subtracted from `uintptr_t` (unsigned integer) value and them it is casted to the `void` pointer.

> In the first image, you can see the issue. I added extra lines for debugging purposes, but the result stays the same. The stk variable holds the value after casting.

I'm not sure how `volatile uint32_t *stk = (void *)*g_idle_topstack;` even compiles when `g_idle_topstack` type is `uintptr_t`. IMHO compiler should report an error there.

emomaxd, 2025-05-15:
I use a custom board in my local so I can't help with that but, nuttx runs on a 32 bit arm cortex r5f core

emomaxd, 2025-05-15:
> Your debug code contains a mistake, in the following line you are dereferencing the top of the stack by adding `*` before casting it to `(void *)` and storing it to `stk` :

I’m aware of that issue — the reason I used the * operator before casting to (void *) and assigning to stk was because I was experimenting with different pointer interactions, such as dereferencing (*) and referencing (&), to better understand the behavior.

The screenshot shows a version where I tried using the * operator, but ultimately it didn’t work as intended. That said, it’s not particularly relevant now, since the original version also didn’t behave correctly.

acassis, 2025-05-17:
@emomaxd could you please explain what you want to accomplish not how you want to do it? Maybe if we understand your goal we could help better. Also since you are using a custom board that you cannot reveal to us (I get it is from TI), please try to duplicate this issue in a common board like the stm32f4discover.

emomaxd, 2025-05-17:
@acassis, 
My goal is to get NuttX running correctly on the R5F core of the TI AM67A SoC—nothing more.

Currently, it crashes at nx_start(), as shown in the attached image.

Some modifications in the code may look unusual (size, stk, ptr) , but they are only there for debugging purposes and to highlight the issue—they’re not part of the intended logic.

The original upstream code doesn’t work on the R5F core. To fix this, I added an & before g_idle_topstack, which made it run correctly.

I've also tested with different compilers, but the issue persists.

At this point, I suspect a possible hardware-related issue, but I’m not sure how to approach it or rule it out.

acassis, 2025-05-17:
@emomaxd thank you for explanation. Could it be something related to cache been enabled? Then when you try to do a right boolean operation the right value is read.

@linguini1 since you did a NuttX port to Raspberry Pi 4 recently, do you have some suggestion that could help @emomaxd here?

linguini1, 2025-05-17:
I am also not too sure here. As others pointed out, `volatile uint32_t *stk = (void *)*g_idle_topstack;` is an odd expression. Can you confirm that `volatile uint32_t *stk = (void *)g_idle_topstack;` also didn't work?

emomaxd, 2025-05-17:
Thanks for your concern, I’ll let you know on Monday.

linguini1, 2025-06-16:
@emomaxd Any updates on this?

linguini1, 2025-06-17:
> @linguini1 I don't work on this anymore, so I can't help with that.

Can the issue be closed then?

acassis, 2025-06-18:
@emomaxd if you can, please submit your port to mainline, it could help more people to experiment and it ends up evolving the support faster