apache/nuttx #16400: [HELP] <title>STM32H723+SDMMC2+IDMA+FATFS,mount failure

snikeguo, 2025-05-16:
### Description

console log:
```
ADE
stm32_sdio_initialize: Initializing SDIO slot 0
stm32_setclkcr: CLKCR: 0003003f PWR: 00000003
stm32_reset: CLCKR: 0003003f POWER: 00000003
stm32_sdio_initialize: Bind SDIO to the MMC/SD driver, minor=0
mmcsd_slotinitialize: minor: 0
mmcsd_hwinitialize: DMA supported: 1
mmcsd_hwinitialize: Attached MMC/SD interrupts
stm32_registercallback: Register 0xc000c965(0xc08084f8)
stm32_callbackenable: eventset: 02
stm32_callback: Callback 0xc000c965(0xc08084f8) cbevents: 02 cdstatus: 00
mmcsd_slotinitialize: MMC/SD slot is empty
mmcsd_slotinitialize: MMC: /dev/mmcsd0 0KB 1-bit backwards compatibility mode
stm32_sdio_initialize: Successfully bound SDIO to the MMC/SD driver
stm32_sdio_initialize: Card detect : 1
sdio_mediachange: cdstatus OLD: 00 NEW: 01
stm32_callback: Callback 0xc000c965(0xc08084f8) cbevents: 02 cdstatus: 01
stm32_callback: Callback to 0xc000c965(0xc08084f8)
mmcsd_mediachange: arg: 0xc08084f8
mmcsd_probe: type: 0 probed: 0
mmcsd_removed: type: 0 present: 1
mmcsd_widebus: No card inserted.
stm32_setclkcr: CLKCR: 0003003f PWR: 00000003
stm32_setclkcr: CLKCR: 0003003f PWR: 00000003
mmcsd_probe: Card present.  Probing....
mmcsd_cardidentify: Identifying card...
stm32_setclkcr: CLKCR: 0003003f PWR: 00000003
stm32_sendcmd: cmd: 00000000 arg: f0f0f0f0 regval: 00001000 enabled irq: 00000000
stm32_sendcmd: cmd: 00000000 arg: 00000000 regval: 00001000 enabled irq: 00000000
stm32_sendcmd: cmd: 00000208 arg: 000001aa regval: 00001108 enabled irq: 00000000
mmcsd_cardidentify: SD V2.x card
stm32_sendcmd: cmd: 00000077 arg: 00000000 regval: 00001137 enabled irq: 00000000
stm32_sendcmd: cmd: 00000129 arg: 40100000 regval: 00001129 enabled irq: 00000000
mmcsd_cardidentify: R3: 40ff8000
stm32_sendcmd: cmd: 00000077 arg: 00000000 regval: 00001137 enabled irq: 00000000
stm32_sendcmd: cmd: 00000129 arg: 40100000 regval: 00001129 enabled irq: 00000000
mmcsd_cardidentify: R3: c0ff8000
mmcsd_cardidentify: SD V2.x card with block addressing
mmcsd_probe: SD version 2.x with block addressing.
stm32_sendcmd: cmd: 000000c2 arg: 00000000 regval: 00001302 enabled irq: 00000000
mmcsd_decode_cid: mid: fe cbx: 0 oid: 32 pnm: SD prv: 0 psn: 0001ed01 mdt: 87         crc: 53
stm32_sendcmd: cmd: 000001c3 arg: 00000000 regval: 00001103 enabled irq: 00000000
mmcsd_sdinitialize: RCA: 0001
stm32_sendcmd: cmd: 0000004d arg: 00010000 regval: 0000110d enabled irq: 00000000
stm32_sendcmd: cmd: 000000c9 arg: 00010000 regval: 00001309 enabled irq: 00000000
mmcsd_decode_csd: CSD:
mmcsd_decode_csd:   CSD_STRUCTURE: 1 SPEC_VERS: 0 (MMC)
mmcsd_decode_csd:   TAAC {TIME_UNIT: 6 TIME_VALUE: 1} NSAC: 0
mmcsd_decode_csd:   TRAN_SPEED {TRANSFER_RATE_UNIT: 2 TIME_VALUE: 6}
mmcsd_decode_csd:   CCC: 1461
mmcsd_decode_csd:   READ_BL_LEN: 9 READ_BL_PARTIAL: 0
mmcsd_decode_csd:   WRITE_BLK_MISALIGN: 0 READ_BLK_MISALIGN: 0
mmcsd_decode_csd:   DSR_IMP: 0
mmcsd_decode_csd:   SD Block Addressing:
mmcsd_decode_csd:     C_SIZE: 29819 SD_ER_BLK_EN: 1
mmcsd_decode_csd:     SD_SECTOR_SIZE: 127 SD_WP_GRP_SIZE: 0
mmcsd_decode_csd:   WP_GRP_EN: 0 MMC DFLT_ECC: 0 (MMC) R2W_FACTOR: 2
mmcsd_decode_csd:   WRITE_BL_LEN: 9 WRITE_BL_PARTIAL: 0
mmcsd_decode_csd:   FILE_FORMAT_GROUP: 0 COPY: 0
mmcsd_decode_csd:   PERM_WRITE_PROTECT: 0 TMP_WRITE_PROTECT: 0
mmcsd_decode_csd:   FILE_FORMAT: 0 ECC: 0 (MMC) CRC: 78
mmcsd_decode_csd: Capacity: 15267840Kb, Block size: 512b, nblocks: 30535680 wrprotect: 0
stm32_sendcmd: cmd: 00000087 arg: 00010000 regval: 00001107 enabled irq: 00000000
stm32_setclkcr: CLKCR: 00031003 PWR: 00000003
stm32_sendcmd: cmd: 00000050 arg: 00000008 regval: 00001110 enabled irq: 00000000
stm32_sendcmd: cmd: 00000077 arg: 00010000 regval: 00001137 enabled irq: 0000012a
[... 193 lines cut ...]
 *                 135 MHz    2
 *                 180 MHz    3
 *                 225 MHz    4
 *  ------------ ---------- -----------
 */

#define BOARD_FLASH_WAITSTATES 3

/* SDMMC definitions ********************************************************/

#define STM32_RCC_D1CCIPR_SDMMCSEL  RCC_D1CCIPR_SDMMC_PLL2

//初始化频率不能高于400K，输入频率为Pll2R=((HSE/PLL2M)*PLL2N)/PLL2R 算出来的值为50Mhz, 50M/(63*2)=396K
#define STM32_SDMMC_INIT_CLKDIV     (63 << STM32_SDMMC_CLKCR_CLKDIV_SHIFT)

#define STM32_SDMMC_MMCXFR_CLKDIV   (3 << STM32_SDMMC_CLKCR_CLKDIV_SHIFT)     //计算方法同上50M/(2*2)=12.5Mhz
#define STM32_SDMMC_SDXFR_CLKDIV    (3 << STM32_SDMMC_CLKCR_CLKDIV_SHIFT)     //计算方法同上50M/(2*2)=12.5Mhz

#define STM32_SDMMC_CLKCR_EDGE      STM32_SDMMC_CLKCR_NEGEDGE

```

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-05-16:
@snikeguo please provide your .config

@JorgeGzm did you try this run STM32H7 this using a configuration similar to this?

snikeguo, 2025-05-16:
@acassis
.config: 
```
#
# Automatically generated file; DO NOT EDIT.
# NuttX/x86_64 Configuration
#

#
# License Setup
#
# CONFIG_ALLOW_BSD_COMPONENTS is not set
# CONFIG_ALLOW_GPL_COMPONENTS is not set
# CONFIG_ALLOW_MIT_COMPONENTS is not set
# CONFIG_ALLOW_ECLIPSE_COMPONENTS is not set
# CONFIG_ALLOW_ICS_COMPONENTS is not set

#
# Build Setup
#
CONFIG_EXPERIMENTAL=y
# CONFIG_DEFAULT_SMALL is not set
CONFIG_HOST_LINUX=y
# CONFIG_HOST_MACOS is not set
# CONFIG_HOST_WINDOWS is not set
# CONFIG_HOST_OTHER is not set

#
# Build Configuration
#
CONFIG_APPS_DIR="../apps"
CONFIG_BASE_DEFCONFIG="ics-stm32h723zg:nsh-dirty"
CONFIG_MODULES=y
CONFIG_BUILD_FLAT=y
# CONFIG_BUILD_PIC is not set
# CONFIG_BUILD_2PASS is not set

#
# Binary Output Formats
#
CONFIG_INTELHEX_BINARY=y
CONFIG_MOTOROLA_SREC=y
CONFIG_RAW_BINARY=y
CONFIG_RAW_DISASSEMBLY=y
# CONFIG_UBOOT_UIMAGE is not set
# CONFIG_DFU_BINARY is not set
CONFIG_OUTPUT_STRIP_PATHS=y

#
# Customize Header Files
#
# CONFIG_ARCH_HAVE_STDINT_H is not set
# CONFIG_ARCH_HAVE_STDBOOL_H is not set
# CONFIG_ARCH_HAVE_MATH_H is not set
# CONFIG_ARCH_FLOAT_H is not set
# CONFIG_DISABLE_FLOAT is not set
CONFIG_ARCH_HAVE_STDARG_H=y
# CONFIG_ARCH_STDARG_H is not set
CONFIG_ARCH_HAVE_SETJMP=y
# CONFIG_ARCH_SETJMP_H is not set
CONFIG_ARCH_NONE_DEBUG_H=y
# CONFIG_ARCH_DEBUG_H is not set
# CONFIG_ARCH_CHIP_DEBUG_H is not set
# CONFIG_ARCH_BOARD_DEBUG_H is not set

#
# Debug Options
#
CONFIG_FORTIFY_SOURCE=0
# CONFIG_NDEBUG is not set
[... 2480 lines cut ...]
# CONFIG_TOOLS_WASM_BUILD is not set

#
# Video Utility libraries
#
# CONFIG_VIDEOUTILS_OPENH264 is not set
# CONFIG_VIDEOUTILS_LIBX264 is not set

#
# Wireless Libraries and NSH Add-Ons
#

#
# Bluetooth applications
#
# CONFIG_NIMBLE is not set

#
# IEEE 802.15.4 applications
#
# CONFIG_IEEE802154_I8SAK is not set
# CONFIG_IEEE802154_LIBMAC is not set
# CONFIG_IEEE802154_LIBUTILS is not set

```

snikeguo, 2025-05-16:
@acassis I've designed a custom development board based on the STM32H723 microcontroller. Since my application runs in SDRAM, I couldn't use the PLL clock configuration functions provided by NuttX. Instead, I configured the PLL clock to the correct values using a J-Link script. This approach shouldn't affect system operation since the OS tick increments at 100 Hz as expected, indicating the system clock is functioning properly. Additionally, SDMMC interrupts are triggering correctly.

My question is: Do I need to configure the CONFIG_FAT_DMAMEMORY macro? I noticed the stm32_dma_alloc_init() function in the codebase, but it's not being called in my implementation. Should this function be explicitly invoked when using SDMMC with DMA on the STM32H723?

snikeguo, 2025-05-17:
solution：

CONFIG_FAT_DMAMEMORY=y
CONFIG_GRAN=y
CONFIG_STM32H7_SDMMC_IDMA=y
CONFIG_ARMV7M_DCACHE_WRITETHROUGH=n see:armv7m mpu document

in stm32_dmarecvsetup:
#if defined(CONFIG_ARCH_HAVE_SDIO_PREFLIGHT)
// DEBUGASSERT(stm32_dmapreflight(dev, buffer, buflen) == 0);
#endif

[image]

snikeguo, 2025-05-17:
```
static sdio_capset_t stm32_capabilities(struct sdio_dev_s *dev)
{
  struct stm32_dev_s *priv = (struct stm32_dev_s *)dev;
  sdio_capset_t caps = 0;

  if (priv->onebit)
    {
      caps |= SDIO_CAPS_1BIT_ONLY;
    }
    else
    {
      caps |=SDIO_CAPS_4BIT_ONLY;  //add this code
    }

  caps |= SDIO_CAPS_DMABEFOREWRITE;

#if defined(CONFIG_STM32H7_SDMMC_IDMA)
  caps |= SDIO_CAPS_DMASUPPORTED;
#endif

  return caps;
}
```

closed 2025-06-01 by snikeguo