apache/nuttx #16409: [FEATURE] Please Support Software Breakpoints

snikeguo, 2025-05-18:
### Is your feature request related to a problem? Please describe.

Some firmware runs in RAM, such as external SDRAM, where breakpoint instructions can be used to implement an unlimited number of breakpoints.

### Describe the solution you'd like

pseudocode:
```
typedef uint16_t platform_bkpt_t;//armv7m bkpt is 16 bits
struct debug_point
{
        int type;
        platform_bkpt_t old_instruction;
       uintptr_t address;
       void *arg;
       debug_callback_t callback;
        bool is_active;
        bool is_software;
}
list<debug_point> debug_points;
int up_debugpoint_add(int type, void *addr, size_t size,
        debug_callback_t callback, void *arg,bool is_software)
{
        struct debug_point dp;
        dp.address = (uintptr_t)addr;
        dp.callback = callback;
        dp.arg = arg;
        dp.is_active = true;
        dp.is_software = is_software;
        dp.type = type;
        
        bool is_valid = false;
        ...
        
        if(is_software)
        {
                is_valid=true;
                memcpy(&dp.old_instruction, (void *)addr, sizeof(platform_bkpt_t));
        }
        else
        {
                // Add hardware breakpoint
                // ...
                var hardware_breakpoint_count=debug_points.select(x=>x.is_software==false).Count();//get count of hardware breakpoints
                if(hardware_breakpoint_count>=MAX_HARDWARE_BREAKPOINTS)
                {
                        // Error: too many hardware breakpoints
                        is_valid=false;
                }
        }
        if(is_valid)
        {
                debug_points.push_back(dp);
        }
        
        return 0;
}
int arm_dbgmonitor(int irq, void *context, void *arg)
{
        var pc=regs[REG_PC];
        var breakpoint=debug_points.find(x=>x.address==pc);
        if(breakpoint!=null)
        {
                if(breakpoint.is_software)
                {
                        // Handle software breakpoint, restore old instruction
                        memcpy((void *)pc, &breakpoint.old_instruction, sizeof(platform_bkpt_t));
                        breakpoint.callback(breakpoint.arg);
                }
                else
                {
                        // Handle hardware breakpoint
                        // ...
                }
                arm_steppoint_match();...
                arm_breakpoint_match()..
                arm_watchpoint_match()...
        }
}
```

### Verification

- [x] I have verified before submitting the report.

xiaoxiang781216, 2025-05-18:
why not add by yourself? if you already has the ideal and demo code.

snikeguo, 2025-05-18:
@xiaoxiang781216  主要是技术能力有限，后续我尝试加一下吧，我的VM虚拟机刚好需要用到这个，现在暂时调试STM32H7 sdmmc读写TF卡太慢的问题。