apache/nuttx #16420: [BUG] Default configuration values hardcoded in code

raiden00pl, 2025-05-21:
### Description / Steps to reproduce the issue

There are many places in NuttX where configuration defaults are set in the source code, not from Kconfig.

This should be handled by Kconfig only otherwise it can lead to hard to detect problems like in this issue https://github.com/apache/nuttx/issues/16256

We can find these definitions using simple commands: 
- `git grep "#define CONFIG"`
- `git grep "  #  define CONFIG"`
- `git grep "  #define CONFIG"`

Some results returned in this way are false positives, so additional verification is required.

In some way the problem is related to this issue https://github.com/apache/nuttx/issues/11637

Another related problem is the use of `#define CONFIG_` for definitions that don't come from Kconfig. Shouldn't definitions in this format be reserved for Kconfig only? When I see `CONFIG_XXX_YYY` definition used somewhere in the code I expect it to be defined in `.config` / `include/nuttx/config.h`, which is often not true in NuttX.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

-

### NuttX Version

-

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Configuring]

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-05-22:
@raiden00pl I think most of those "failsafe" #ifndef / #define comes from an era when NuttX didn't use Kconfig. Since Kconfig introduction it doesn't make sense anymore.

xiaoxiang781216, 2025-05-22:
if these macros can't migrate to Kconfig soon, it's better to remove CONFIG_ prefix and #ifndef check to avoiding confuse people.

acassis, 2025-05-22:
@xiaoxiang781216 I think one downside of removing CONFIG_ is that these macros may become hard to find later. It's better to remove them step by step, adding them to Kconfig.