apache/nuttx #16429: [BUG] nat_port_select() returned value not safe

CosmicRoach, 2025-05-23:
### Description / Steps to reproduce the issue

Returned Value:
 *   External port number on success; 0 on failure
In some cases，when the return value is 0, it may indicate success.

[image]

ipv4_nat_outbound_icmp() use icmp->id as port, but some careless applications may send ICMP request packets with icmp->id == 0, in this case, nat_port_select() will return a port number of 0, this 0 is a selected result, not a failure to find a port. 

[image]

The return value of 0 is interpreted as an error here, preventing subsequent processes from proceeding.
[image]

### On which OS does this issue occur?

[OS: Other]

### What is the version of your OS?

.

### NuttX Version

all

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Networking]

### Verification

- [x] I have verified before submitting the report.

xiaoxiang781216, 2025-05-24:
@wengzhe could you look at this problem?

CosmicRoach, 2025-05-27:
> [@zhhyu7](https://github.com/zhhyu7) I think we have two ways to fix it, which way do you think is better?
> 
> 1. Don't allow choosing `icmp->id` as `0`, that is, return a positive number when `local_port == 0`.
> 2. Change the return value type, and use a negative number as an error.

I believe the second way is better. I'm currently working on a router project using NuttX. During debugging on a PC with Wireshark, I observed that the `icmp->id`  between certain clients and servers are indeed 0. However, as a router, we cannot simply drop these packets.

CosmicRoach, 2025-05-29:
> [@CosmicRoach](https://github.com/CosmicRoach) could you create PR to fix it?

Okay, I will create a PR.

casaroli, 2026-11-08:
Fixed by #18812.

closed 2026-11-08 by casaroli