apache/nuttx #16432: [FEATURE] Nuttx Documentation improvemet

abisop, 2025-05-23:
### Is your feature request related to a problem? Please describe.

# Documentation improvement

### Describe the solution you'd like

# I am proposing to use Markdown for documentation.

## Why Markdown : 
1. Its easy to use in daily routine 
2.  Supported by modern js frameworks (In this case -> vitepress)
3. We can leverage many graphical UI features using pt. 2 . (Inspiration - PX4-Autopilot)

Here i am attaching source and website link :
[source](https://github.com/abisop/nuttx-docs.git)
[Experimental Website](https://abisop.github.io/)

### Verification

- [x] I have verified before submitting the report.

simbit18, 2025-05-23:
https://github.com/apache/nuttx/issues/15893

abisop, 2025-05-23:
@simbit18 I am not a web developer. But the process of static site generator + some custom  automation is such that Now we have to write only markdown files and rest is taken care of.

raiden00pl, 2025-05-23:
-1. Spinx was chosen a long time ago and was just a replacement for Markdown and HTML. Going back to Markdown is unnecessary work that adds nothing to this project. 
Here is the topic on the mailing list: https://lists.apache.org/thread/25fb57pg1kdcjrv4m6vm4qhl8x225hsc

linguini1, 2025-05-23:
I would also vote -1 on this. Changing to Markdown doesn't feel like it brings any major advantages over RST, and would come with a ton of work moving over the existing documentation system. We haven't even managed to convert all the old README.txt files yet (#11077)

abisop, 2025-05-23:
> -1. Spinx was chosen a long time ago and was just a replacement for Markdown and HTML. Going back to Markdown is unnecessary work that adds nothing to this project. Here is the topic on the mailing list: https://lists.apache.org/thread/25fb57pg1kdcjrv4m6vm4qhl8x225hsc

@raiden00pl Thanks for pointing... 
I found that sir @gregory-nutt used to write docs in html which followed by use of reStructuredText. But markdown is different as we all know .
We are not going back but with the trend where frameworks of js like vitepress are used for creating modular static websites by consuming markdown files.

abisop, 2025-05-23:
@raiden00pl @linguini1 
I think writing markdown is easy and effective. So that new people can easily edit without fuzz
Also js based static site generators use it.

I am not here for rst vs markdown. You guys know better what is what...
I have done my homework and this was my first assignment to improve docs please let me know...
> [source](https://github.com/abisop/nuttx-docs.git)
[Experimental Website](https://abisop.github.io/)  

You can try [edit on github]() on bottom of every page 💯

raiden00pl, 2025-05-24:
@abisop We currently use several features in Sphinx that are not available by default in Markdown like cross-referencing or TOC generation, so this is not so simple migration even if we agree to this.
Sphinx is much more useful for creating complex documentation for such large projects like NuttX. If you look at the two functionally closest projects to NuttX, which are Zephyr and Linux, both use Sphinx for documentation. The same for mynewt, rterms and probably more.

abisop, 2025-05-25:
@acassis @raiden00pl  Should we add "edit on github" option. 
this will allow people to edit easily. So if someone finds small typo they will be able to add PR for change with ease

raiden00pl, 2025-05-25:
`mySt` is just a workaround to support legacy documentation format that is slowly being migrated to rst. Once everything is migrated, it should be removed.

michallenc, 2025-05-25:
Jumping a bit late to the discussion, but where do we stand with autogenerated API documentation? Pages like [this](https://nuttx.apache.org/docs/latest/components/drivers/character/analog/adc/index.html) would benefit from having doxygen (or else) comments for structures, enums, defines, etc.

raiden00pl, 2025-05-25:
As far as I remember Gregory was always against cluttering the code with metadata comments like Doxygen. Personally I agree with this view. The code should first be readable by users, not generators. Unfortunately this makes auto-generating API documentation not so easy.

michallenc, 2025-05-25:
> As far as I remember Gregory was always against cluttering the code with metadata comments like Doxygen. Personally I agree with this view. The code should first be readable by users, not generators. Unfortunately this makes auto-generating API documentation not so easy.

That's a fair point, but we already have comments in the code (include large blocks for functions). There isn't much difference in readability between, let's say.

```
begin_packed_struct struct adc_msg_s
{
  uint8_t      am_channel;               /* The 8-bit ADC Channel */
  int32_t      am_data;                  /* ADC convert result (4 bytes) */
} end_packed_struct;
```

and

```
begin_packed_struct struct adc_msg_s
{
  /*!  The 8-bit ADC Channel */
  uint8_t      am_channel;
  /*! ADC convert result (4 bytes) */
  int32_t      am_data;
} end_packed_struct;
```

Yes, it can get a bit messy when referencing other structures, functions, members, etc (`@ref xyz` etc). But maybe the public API is not as critical for automatic generation, because we don't expect it to change.