apache/nuttx #16462: [BUG] nxsem_wait_irq assertion failure

atr294, 2025-05-30:
### Description / Steps to reproduce the issue

* Description:

When running a time-consuming task in the background and repeatedly entering any rando keys in the NSH shell, a crash sometimes happens in the `nxsem_wait_irq()` function in `sem_waitirq.c`:
```bash
nsh> [  416.373084] dump_assert_info: Current Version: NuttX  12.9.0 29c4c87e2c-dirty May 30 2025 12:48:55 risc-v
[  416.373113] dump_assert_info: Assertion failed : at file: semaphore/sem_waitirq.c:82 task: Idle_Task process: Kernel 0x80200f62
[  416.373156] up_dump_register: EPC: 802135a2
[  416.373175] up_dump_register: A0: 806067b0 A1: 00000052 A2: 00000000 A3: 00000000
[  416.373209] up_dump_register: A4: 00000006 A5: 00000007 A6: 98968000 A7: 2e24e000
[  416.373245] up_dump_register: T0: 80605818 T1: 00000020 T2: 00001000 T3: 00000028
[  416.373280] up_dump_register: T4: 80045f40 T5: 00000074 T6: 00000000
```

```c
DEBUGASSERT(sem != NULL && atomic_read(NXSEM_COUNT(sem)) < 0);  // In fact the issue is with DEBUGASSERT(atomic_read(NXSEM_COUNT(sem)) < 0))
```
* Platforms affected: RISC-V QEMU on knsh, RISC-V Litex
* To reproduce:

1. Run `ostest &`
2. Keep typing anything randomly and pressing enter in the shell, the DEBUGASSERT can be triggered within a few seconds.
3. This issue started at commit `510d9695` (`sem_waitirq: Use kmap interface to access the semaphore`). It did not happen in earlier commits.

Note: `nuttx-apps` should be at around commit `2cfbdbb40` to be able to build the QEMU project. I used this command to start qemu:
`qemu-system-riscv32 -semihosting -M virt,aclint=on -cpu rv32 -kernel nuttx -nographic`

* Temporary fix:

It looks like there is a race condition: after the task starts waiting for the semaphore, but before a timeout or signal happens, another thread or interrupt might post (release) the semaphore. This means the semaphore could become available just before the timeout or signal handler runs.

To work around this, I added extra checks before calling `nxsem_wait_irq()` in `sem_timeout.c` and `sig_dispatch.c`:

```c
if (stcb->task_state == TSTATE_WAIT_SEM)
    {
      FAR sem_t *sem = stcb->waitobj;

      DEBUGASSERT(sem != NULL);

      if (atomic_read(NXSEM_COUNT(sem)) < 0)
        {
          nxsem_wait_irq(stcb, ETIMEDOUT);
        }
      else
        {
          /* If the semaphore was not taken, then we can just return
           * without changing the task state.
           */

          stcb->waitobj    = NULL;

          /* Add the task to the ready-to-run list */

          if (nxsched_add_readytorun(stcb))
            {
              up_switch_context(stcb, this_task());
            }
        }
    }
```

This seems to solve the immediate problem, but will cause other crash at sem_wait.c such as:
```bash
[2666886928007168.001000] dump_assert_info: Current Version: NuttX  12.9.0 29c4c87e2c-dirty May 30 2025 13:02:49 risc-v
[2670275657203712.001000] dump_assert_info: Assertion failed : at file: semaphore/sem_wait.c:164 task: ostest process: ostest 0xc0008226
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Arch Linux

### NuttX Version

12.9.0

### Issue Architecture

[Arch: risc-v]

### Issue Area

[Area: Kernel]

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-05-31:
@pussuw could you please take a look, it is related to your PR about sem_waitirq

acassis, 2025-05-31:
@atr294 nice test idea! Did you try to run ostest in background and the monkey on foreground to see if the issue also happens?

atr294, 2025-06-02:
@acassis thanks for the suggestion. I've tried the monkey test in the foreground, and it doesn't seem to trigger the same issue. 

Running `monkey -t 0x12 ` or `monkey -t 0x11` in the foreground, while running `ostest &' in the background only caused this non-crash exception:
`[   29.897673] riscv_exception: EXCEPTION: Store/AMO page fault. MCAUSE: 0000000f, EPC: 80208daa, MTVAL: 80c00000
[   29.898245] riscv_exception: Segmentation fault in PID 2: /system/bin/init`

My understanding is monkey app doesn't seem to support keyboard input, which is closer to my test condition. It only supports the touch and button input. 

I'm not sure the above exception worth looking into, but running them without ostest in the background doesn't cause any issue.

`nsh> monkey -t 0x12
[241634860072960.001000] [monkey] monkey_dev_create: open /dev/ubutton success, fd = 3, type: ubutton
[244271969992704.001000] [monkey] monkey_create: OK
[244800250970112.001000] [monkey] monkey_init: Screen: 480x480, offset: 0,0
[245246927568896.001000] [monkey] monkey_init: Period: 100 ~ 500ms
[245706489069568.001000] [monkey] monkey_init: Button bit: 0
[246062971355136.001000] [monkey] monkey_init: Log level: 1
[246428043575296.001000] [monkey] monkey_init: Event 0(click): weight=70 duration 50 ~ 200ms
[246921964814336.001000] [monkey] monkey_init: Event 1(long-press): weight=10 duration 400 ~ 600ms
[247428770955264.001000] [monkey] monkey_init: Event 2(drag): weight=20 duration 100 ~ 400ms
`

pussuw, 2025-06-02:
Duplicate of https://github.com/apache/nuttx/issues/8917

The underlying issue is that when nxsem_wait_irq is called, the wrong mappings might be active. In order for it to work properly, CONFIG_MM_KMAP needs to be enabled. I see that none of the rv-virt targets have it.

atr294, 2025-06-11:
@pussuw @acassis My apologies for the delay.

It took me some time to get `CONFIG_MM_KMAP` working on my Litex target due to initial memory layout issues. After resolving those, I can confirm that enabling `CONFIG_MM_KMAP` fixes the problem. The system is now stable and works correctly on both QEMU and Litex.

I've opened a PR for the `qemu-rv/rv-virt:knsh` configuration:

[(https://github.com/apache/nuttx/pull/16508)]

Thanks very much for the help!

closed 2025-06-12 by acassis