apache/nuttx #16504: [HELP] Why isn't multicast loopback switched on by default?

sastel, 2025-06-10:
### Description

@acassis I noticed recently that multicast loopback is disabled by default in NuttX. I have a pair of applications which when I run them on the same device using Linux, discover each other via multicast loopback. However when I run these applications on the same device using NuttX (kernel mode), they do not discover each other, even though applications on other devices can see their multicast traffic.

In `nuttx/net/udp/udp_send.c`, you can see that there is support for `IP_MULTICAST_LOOP` in the code. I have the relevant options set in my defconfig, so if that bit is set then it should be looping back.

It seems that `IP_MULTICAST_LOOP` is set by default in Linux, but in NuttX the bit is not set by default in `conn->sconn.s_options`.

I have two questions based on this, and I'm hoping someone more knowledgeable than I about this stuff can help.
1. Why doesn't NuttX set this bit by default?
2. Should NuttX set this bit by default?

I can use the sockets API to set this in my communications stack, but it isn't clear to me what is the right place to do it.

### Verification

- [x] I have verified before submitting the report.

xiaoxiang781216, 2025-06-10:
POSIX-compliance isn't an option. It's also bad to add an option to control the minor default behavior, since only few people can remember them.

TimJTi, 2025-06-10:
Ok, but I see people lamenting the loss of options that allow NuttX to better suit small processors. Shouldn't we accommodate that somehow as NON default options?