apache/nuttx #16539: [BUG] Windows: Cake sim:windows fail to build

simbit18, 2025-06-16:
### Description / Steps to reproduce the issue

cmake -B build -DBOARD_CONFIG=sim:windows -G"Visual Studio 17 2022" -A Win32

cmake --build build

```
mm.vcxproj -> D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\mm\Debug\mm.lib
  net.vcxproj -> D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\net\Debug\net.lib
  group_create.c
  group_join.c
  nx_bringup.c
  instrument.c
  reboot_notifier.c
D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\sched\wqueue\wqueue.h(91,25): error C2229: struct 'hp_wqueue_s' has an illegal zero-sized array [D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\sched\sched.vcxproj]
  (compiling source file '../../sched/init/nx_bringup.c')
  
D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\sched\wqueue\wqueue.h(106,25): error C2229: struct 'lp_wqueue_s' has an illegal zero-sized array [D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\sched\sched.vcxproj]
  (compiling source file '../../sched/init/nx_bringup.c')
  
  mq_initialize.c
  pthread_exit.c
  pthread_join.c
  pthread_mutex.c
  pthread_mutexconsistent.c
  sched_foreach.c
  sched_lock.c
  sem_timedwait.c
  sem_clockwait.c
  sig_releasependingsigaction.c
  sig_unmaskpendingsignal.c
  task_start.c
  task_delete.c
  timer_delete.c
  kwork_queue.c
D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\sched\wqueue\wqueue.h(91,25): error C2229: struct 'hp_wqueue_s' has an illegal zero-sized array [D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\sched\sched.vcxproj]
  (compiling source file '../../sched/wqueue/kwork_queue.c')
  
D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\sched\wqueue\wqueue.h(106,25): error C2229: struct 'lp_wqueue_s' has an illegal zero-sized array [D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\sched\sched.vcxproj]
  (compiling source file '../../sched/wqueue/kwork_queue.c')
  
  Generating Code...
  Compiling...
  kwork_cancel.c
D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\sched\wqueue\wqueue.h(91,25): error C2229: struct 'hp_wqueue_s' has an illegal zero-sized array [D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\sched\sched.vcxproj]
  (compiling source file '../../sched/wqueue/kwork_cancel.c')
  
D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\sched\wqueue\wqueue.h(106,25): error C2229: struct 'lp_wqueue_s' has an illegal zero-sized array [D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\sched\sched.vcxproj]
  (compiling source file '../../sched/wqueue/kwork_cancel.c')
  
  kwork_thread.c
D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\sched\wqueue\wqueue.h(91,25): error C2229: struct 'hp_wqueue_s' has an illegal zero-sized array [D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\sched\sched.vcxproj]
  (compiling source file '../../sched/wqueue/kwork_thread.c')
  
D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\sched\wqueue\wqueue.h(106,25): error C2229: struct 'lp_wqueue_s' has an illegal zero-sized array [D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\sched\sched.vcxproj]
  (compiling source file '../../sched/wqueue/kwork_thread.c')
  
  Generating Code...
  An error occurred while copying files: Cannot find path 'D:\a\nuttx_test_pr\nuttx_test_pr\sources\nuttx\build\Debug\nuttx.exe' because it does not exist.
```

### On which OS does this issue occur?

[OS: Windows]

### What is the version of your OS?

GitHub Actions

### NuttX Version

master

### Issue Architecture

[Arch: simulator]

### Issue Area

[Area: Build System]

### Verification

- [x] I have verified before submitting the report.

xiaoxiang781216, 2025-06-16:
@lupyuen could we restore sim:windows ci to catch this type of error in the future?

acassis, 2025-06-16:
@simbit18 what is the value of CONFIG_SCHED_HPNTHREADS ?

simbit18, 2025-06-17:
Hi @xiaoxiang781216 @acassis @anchao  

This PR #16231 
 **sched/wqueue: Refactor delayed and periodic workqueue**

broke the build of sim:windows.

simbit18, 2025-06-17:
> [@simbit18](https://github.com/simbit18) what is the value of CONFIG_SCHED_HPNTHREADS ?

Hi @acassis 

```
# Work queue support
#
CONFIG_SCHED_WORKQUEUE=y
# CONFIG_WQUEUE_NOTIFIER is not set
CONFIG_SCHED_HPWORK=y
CONFIG_SCHED_HPNTHREADS=1
CONFIG_SCHED_HPWORKPRIORITY=224
CONFIG_SCHED_HPWORKSTACKSIZE=2048
CONFIG_SCHED_LPWORK=y
CONFIG_SCHED_LPNTHREADS=1
CONFIG_SCHED_LPWORKPRIORITY=100
CONFIG_SCHED_LPWORKSTACKSIZE=2048
# end of Work queue support
```

[config.txt](https://github.com/user-attachments/files/20773587/config.txt)

acassis, 2025-06-17:
> Hi [@xiaoxiang781216](https://github.com/xiaoxiang781216) [@acassis](https://github.com/acassis) [@anchao](https://github.com/anchao)
> 
> This PR [#16231](https://github.com/apache/nuttx/pull/16231) **sched/wqueue: Refactor delayed and periodic workqueue**
> 
> broke the build of sim:windows.

This is why I added the tag #needsmoretesting, no surprises here!

xiaoxiang781216, 2025-06-17:
@Fix-Point please look at this build break.

Fix-Point, 2025-06-18:
Please check if this https://github.com/apache/nuttx/pull/16545 fixed this.

xiaoxiang781216, 2025-06-18:
> 
> I am trying to understand why it does not show error (error stream) on our CI !!!

@simbit18 do you have any finding? we need make ci check work to avoid the break in the future.

simbit18, 2025-06-18:
> > I am trying to understand why it does not show error (error stream) on our CI !!!
> 
> [@simbit18](https://github.com/simbit18) do you have any finding? we need make ci check work to avoid the break in the future.

Hi @xiaoxiang781216 yes I am testing

simbit18, 2025-06-18:
> > I am trying to understand why it does not show error (error stream) on our CI !!!
> 
> [@simbit18](https://github.com/simbit18) do you have any finding? we need make ci check work to avoid the break in the future.

Hi @xiaoxiang781216 PR https://github.com/apache/nuttx/pull/16546