apache/nuttx #16555: [BUG] Unix datagram sockets seems broken

michallenc, 2025-06-19:
### Description / Steps to reproduce the issue

Unix local datagram sockets seems to be broken on NuttX. The following code doesn't work on `sim:udgram` configuration (nor other configurations with `CONFIG_NET=y`, including real hardware). The code basically creates a local unix socket and then polls for `POLLIN`. However, the poll fails on `POLLERR`, but no `errno` is set during subsequent `read` call (and `read` returns 0). I think the POSIX states that subsequent `read` should return `-1` and set `errno` in this case. You can compile it on both Linux and NuttX to reproduce, just run the application without argument (or with `'s'`) for server and with `'c'` for a client). The code works on Linux as expected - server polls, client sends 10 and server receives and prints the value. No `POLLERR` occurs.

```c
#include <errno.h>
#include <poll.h>
#include <stdbool.h>
#include <stdint.h>
#include <stdio.h>
#include <sys/socket.h>
#include <sys/un.h>
#include <unistd.h>

#define SOCKET_PATH "/tmp/socket.sock"

void run_client(void) {
  int sock = socket(AF_UNIX, SOCK_DGRAM, 0);
  if (sock < 0) {
    printf("Failed to open socket.");
    return;
  }

  struct sockaddr_un name;
  name.sun_family = AF_UNIX;
  strcpy(name.sun_path, SOCKET_PATH);

  uint8_t data = 10;
  if (sendto(sock, &data, sizeof data, 0, (const struct sockaddr *)&name,
             sizeof name) < 0) {
    printf("Failed to send data.");
  }

  close(sock);
}

void run_server(void) {
  int sock = socket(AF_UNIX, SOCK_DGRAM, 0);
  if (sock == -1) {
    printf("Failed to create a socket: %d", errno);
    return;
  }

  struct sockaddr_un addr;
  memset(&addr, 0, sizeof addr);
  addr.sun_family = AF_UNIX;
  strcpy(addr.sun_path, SOCKET_PATH);

  if (bind(sock, (const struct sockaddr *)&addr, sizeof addr) == -1) {
    printf("Failed to bind a socket: %d", errno);
    close(sock);
    return;
  }

  struct pollfd pfd = {.fd = sock, .events = POLLIN};

  uint8_t data;
  while (true) {
    int ret = poll(&pfd, 1, 10000);
    if (ret == -1 || ret == 0) {
      printf("timeout\n");
      continue;
    } else if (pfd.revents & POLLIN) {
      read(sock, &data, sizeof data);
      printf("data = %d\n", data);
    } else if (pfd.revents & POLLERR) {
      read(sock, &data, sizeof data);
      printf("pollerr = %d\n", errno);
      break;
    }
[... 397 lines cut ...]
  CONFIG_NSH_FILEIOSIZE=1024
  CONFIG_NSH_SYSINITSCRIPT=\init.d/rc.sysinit\
  CONFIG_NSH_INITSCRIPT=\init.d/rcS\
  CONFIG_NSH_SCRIPT_REDIRECT_PATH=\\
  CONFIG_NSH_CONSOLE=y
  CONFIG_NSH_ARCHINIT=y
  CONFIG_NSH_NETINIT=y
  CONFIG_NSH_WGET_BUFF_SIZE=512
  CONFIG_SYSTEM_DD=y
  CONFIG_SYSTEM_DD_PROGNAME=\dd\
  CONFIG_SYSTEM_DD_PRIORITY=100
  CONFIG_SYSTEM_DD_STACKSIZE=2048
  CONFIG_SYSTEM_DD_STATS=y
  CONFIG_SYSTEM_NSH=y
  CONFIG_SYSTEM_NSH_PRIORITY=100
  CONFIG_SYSTEM_NSH_STACKSIZE=2048
  CONFIG_SYSTEM_NSH_PROGNAME=\nsh\
  CONFIG_SYSTEM_READLINE=y
  CONFIG_READLINE_HAVE_EXTMATCH=y
  CONFIG_READLINE_ECHO=
```

### Verification

- [x] I have verified before submitting the report.

michallenc, 2025-06-19:
So it seems there are two problems. The first one is pretty obvious after looking to the code, we simply don't support `poll` on `SOCK_DGRAM` 😅 See https://github.com/apache/nuttx/blob/master/net/local/local_netpoll.c#L162

The other problem is in `local_recvmsg` and `local_sendmsg` in the first check

```
  if (conn->lc_outfile.f_inode == NULL)
    {
      return -EPIPE;
    }
```

added in #14103. For some reason, this seems to be always NULL for datagram in our case. With few hacks, I was able to send/receive data with `SOCK_DGRAM`, but the behavior is still weird.

acassis, 2025-06-23:
@xiaoxiang781216 @wengzhe could you please comment here?

michallenc, 2025-07-06:
@xiaoxiang781216 is @GUIDINGLI actively working on this? I would at least merge the `conn->lc_infile.f_inode == NULL` for now, to at least get datagrams working. Polling can be solved later.