apache/nuttx #16570: [BUG] CONFIG_FILE_STREAM needs to be enabled by default for some applications (i.e. can example)

acassis, 2025-06-22:
### Description / Steps to reproduce the issue

I was getting this error after enabling CAN example:

LD: nuttx
arm-none-eabi-ld: /home/alan/nuttxspace/nuttx/staging/libapps.a(can_main.c.home.alan.nuttxspace.apps.examples.can_1.o): in function `show_usage':
can_main.c:(.text.show_usage+0x6): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0xe): undefined reference to `fprintf'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x14): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x1c): undefined reference to `fprintf'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x22): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x2a): undefined reference to `fputs'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x30): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x38): undefined reference to `fputs'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x3e): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x46): undefined reference to `fputs'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x4c): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x54): undefined reference to `fputs'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x5a): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x64): undefined reference to `fprintf'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x6a): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.show_usage+0x76): undefined reference to `fputs'
arm-none-eabi-ld: /home/alan/nuttxspace/nuttx/staging/libapps.a(can_main.c.home.alan.nuttxspace.apps.examples.can_1.o): in function `can_main':
can_main.c:(.text.can_main+0x54): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.can_main+0x84): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.can_main+0x8c): undefined reference to `fputs'
arm-none-eabi-ld: can_main.c:(.text.can_main+0xb0): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.can_main+0xce): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.can_main+0xda): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.can_main+0x122): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.can_main+0x12a): undefined reference to `fputs'
arm-none-eabi-ld: can_main.c:(.text.can_main+0x16e): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.can_main+0x172): undefined reference to `fflush'
arm-none-eabi-ld: can_main.c:(.text.can_main+0x1da): undefined reference to `lib_get_stream'
arm-none-eabi-ld: can_main.c:(.text.can_main+0x1de): undefined reference to `fflush'

It was fixed after enabling CONFIG_FILE_STREAM=y

### On which OS does this issue occur?

[OS: Other], [OS: Linux]

### What is the version of your OS?

Ubuntu 24.04

### NuttX Version

mainstream

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Applications]

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-06-22:
@xiaoxiang781216 I think we need to verify which applications will need to select CONFIG_FILE_STREAM automatically to avoid issues like this

TimJTi, 2025-06-22:
>...BTW, the better fix is switching fopen/fclose to open/close as much as we can.

I thought fopen/fclose was the preferred, portable, method? Since open/close are not ANSI C IIRC.

xiaoxiang781216, 2025-06-23:
> > ...BTW, the better fix is switching fopen/fclose to open/close as much as we can.
> 
> I thought fopen/fclose was the preferred, portable, method? Since open/close are not ANSI C IIRC.

open/close is specified by POSIX. So, if you don't plan to port your code to Windows, open/close is equal to fopen/fclose.

TimJTi, 2025-06-23:
> open/close is specified by POSIX. So, if you don't plan to port your code to Windows, open/close is equal to fopen/fclose.

Apologies - most of my (patchy!) software life has been "pure" C not POSIX so I probably remembered something from back way yonder that, in this/our context, was incorrect. Sorry!

acassis, 2025-06-26:
> > [@xiaoxiang781216](https://github.com/xiaoxiang781216) I think we need to verify which applications will need to select CONFIG_FILE_STREAM automatically to avoid issues like this
> 
> not select, but depend on CONFIG_FILE_STREAM. BTW, the better fix is switching fopen/fclose to open/close as much as we can.

I think the CAN example is already using open/close instead of fopen/fclose, the issue there is because it is using fprintf().

acassis, 2025-06-28:
CAN app fixed!

closed 2025-06-28 by acassis