apache/nuttx #16610: [FEATURE] Rp2040 Rp23xx Disable Spi PrimeCell SSP peripheral before to reconfigure parameters

paolovolpi, 2025-06-25:
### Is your feature request related to a problem? Please describe.

As done by Pico Sdk would be better to unset SSPCR1.SSE bit  before to modify Spi parameters (Mode, Frequency and Bits).

https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/hardware_spi/spi.c#L47

### Describe the solution you'd like

In functions spi_setfrequency, spi_setmode, spi_setbits before to call spi_putreg on  SSPCR0 SSPCR1 registries disable the PrimeCell SSP peripheral:
```
regval = spi_getreg(priv, RP23XX_SPI_SSPCR1_OFFSET);
regval &= ~RP23XX_SPI_SSPCR1_SSE;
spi_putreg(priv, RP23XX_SPI_SSPCR1_OFFSET, regval);
```
After the SSPCR0 SSPCR1 registries modification, re-enable  the PrimeCell SSP peripheral:
```
regval = spi_getreg(priv, RP23XX_SPI_SSPCR1_OFFSET);
regval |= RP23XX_SPI_SSPCR1_SSE;
spi_putreg(priv, RP23XX_SPI_SSPCR1_OFFSET, regval);
```

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-06-26:
@paolovolpi please feel free to submit a patch with this modification.

shtirlic, 2025-07-09:
PR by @paolovolpi  https://github.com/apache/nuttx/pull/16704

shtirlic, 2025-08-08:
could be closed since #16704 merged

closed 2025-08-08 by xiaoxiang781216