apache/nuttx #16689: [HELP] ps crashes on second call

comejv, 2025-07-07:
### Description

Hello,
I am compiling NuttX for an STM32H753ZI and I have issues with getting `ps` to work.
Relevant options in my `.config` (full config bottom of message):
```
CONFIG_FS_PROCFS=y
CONFIG_FS_PROCFS_REGISTER=y
CONFIG_NSH_PROC_MOUNTPOINT="/proc"
```
Compiled with CMake and the `nucleo-h743zi:nsh` target.

I then flash with openocd and open picocom on /dev/ttyACM0 to get an nsh prompt.

First thing weird, proc is not mounted? Should it be mounted automatically by stm32_bringup.c here:
```c
#ifdef CONFIG_FS_PROCFS
  /* Mount the procfs file system */

  ret = nx_mount(NULL, STM32_PROCFS_MOUNTPOINT, "procfs", 0, NULL);
  if (ret < 0)
    {
      syslog(LOG_ERR,
             "ERROR: Failed to mount the PROC filesystem: %d\n",  ret);
    }
#endif /* CONFIG_FS_PROCFS */
```

So I mount it manually with `mount -t procfs /proc` and then run:
```shell
nsh> usleep 10000000 &
sh [2:100]
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK            STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 0001008 Idle_Task
    1     1 100 RR       Task      - Running            0000000000000000 0002008 nsh_main
    2     2 100 RR       Task      - Waiting  Signal    0000000000000000 0001992 sh -c usleep
nsh> ps
  PID GROUP PRI dump_assert_info: Current Version: NuttX  0.0.0 1df6aa92-dirty Jan  1 1980 00:00:00 arm
dump_assert_info: Assertion failed panic: at file: :0 task: nsh_main process: nsh_main 0x800cf51
up_dump_register: R0: 00000610 R1: 380009a4 R2: 38000518  R3: 00000610
up_dump_register: R4: 38001fa8 R5: 0802220c R6: 00000000  FP: 00000000
up_dump_register: R8: 00000000 SB: 00000000 SL: 00000000 R11: 00000000
up_dump_register: IP: 00000000 SP: 380009d0 LR: 080063e5  PC: 0800e920
up_dump_register: xPSR: 61000000 BASEPRI: 00000000 CONTROL: 00000000
up_dump_register: EXC_RETURN: ffffffe9
dump_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK          STACKBASE  STACKSIZE   COMMAND
dump_task:       0     0   0 FIFO     Kthread -   Ready              0000000000000000 0x240018b4      1008   Idle_Task
dump_task:       1     1 100 RR       Task    -   Running            0000000000000000 0x38000518      2008   nsh_main
dump_task:       2     2 100 RR       Task    -   Waiting Signal     0000000000000000 0x380014a8      1992   sh -c usleep
```

Why did it crash the second time?

I tried changing `CONFIG_TASK_NAME_SIZE` with no success.
I tried increasing `CONFIG_SYSTEM_NSH_STACKSIZE` to 4096 as mentioned in #11463 with no avail, only change was reported stack size in the crash:
```
dump_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK          STACKBASE  STACKSIZE   COMMAND
dump_task:       0     0   0 FIFO     Kthread -   Ready              0000000000000000 0x240018b4      1008   Idle_Task
dump_task:       1     1 100 RR       Task    -   Running            0000000000000000 0x38000518      2008   nsh_main
dump_task:       2     2 100 RR       Task    -   Waiting Signal     0000000000000000 0x380014a8      4040   sh -c usleep
```

Any help greatly appreciated.

Full `.config`:
<details>
  <summary>Click me</summary>
  CONFIG_EXPERIMENTAL=y
CONFIG_HOST_LINUX=y
[... 264 lines cut ...]
CONFIG_NSH_CODECS_BUFSIZE=128
CONFIG_NSH_CMDOPT_HEXDUMP=y
CONFIG_NSH_PROC_MOUNTPOINT="/proc"
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_SYSINITSCRIPT="init.d/rc.sysinit"
CONFIG_NSH_INITSCRIPT="init.d/rcS"
CONFIG_NSH_SCRIPT_REDIRECT_PATH=""
CONFIG_NSH_CONSOLE=y
CONFIG_SYSTEM_DD=y
CONFIG_SYSTEM_DD_PROGNAME="dd"
CONFIG_SYSTEM_DD_PRIORITY=100
CONFIG_SYSTEM_DD_STACKSIZE=2048
CONFIG_SYSTEM_DD_STATS=y
CONFIG_SYSTEM_NSH=y
CONFIG_SYSTEM_NSH_PRIORITY=100
CONFIG_SYSTEM_NSH_STACKSIZE=2048
CONFIG_SYSTEM_NSH_PROGNAME="nsh"
CONFIG_SYSTEM_READLINE=y
CONFIG_READLINE_HAVE_EXTMATCH=y
CONFIG_READLINE_ECHO=y
</details>

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-07-08:
@comejv maybe there is some issues in your config, I tested here using weact-stm32h743 and the issue didn't show up:

```
NuttShell (NSH) NuttX-12.9.0                                                 
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK            STACK D
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 0001008 >
    1     0 224 RR       Kthread   - Waiting  Semaphore 0000000000000000 0001984 0
    3     3 100 RR       Task      - Running            0000000000000000 0002008 >
nsh> usleep 10000000 &
sh [4:100]                                                                   
nsh> uname -a
NuttX 12.9.0 541450e692-dirty Jul  3 2025 17:19:34 arm weact-stm32h743            
nsh> ps                                                                           
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK            STACK D
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 0001008 >
    1     0 224 RR       Kthread   - Waiting  Semaphore 0000000000000000 0001984 0
    3     3 100 RR       Task      - Running            0000000000000000 0002008 >
    4     4 100 RR       Task      - Waiting  Signal    0000000000000000 0001992 p
nsh> ps                                                                           
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK            STACK D
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 0001008 >
    1     0 224 RR       Kthread   - Waiting  Semaphore 0000000000000000 0001984 0
    3     3 100 RR       Task      - Running            0000000000000000 0002008 >
    4     4 100 RR       Task      - Waiting  Signal    0000000000000000 0001992 p
nsh> 
```

acassis, 2025-07-08:
@comejv looking your config I didn't see nothing that highlight a configuration issue. Initially I thought you could have a small stack memory definition, but you are using the double of my stack (4KB vs 2KB).

Probably it could be some issue with toolchain or something related to out-of-tree custom board. Could you please try to setup same configuration but using the "in-tree" boards/arm/stm32h7/nucleo-h743zi (or nucleo-h743zi2 depending on your board version) ?

Also please provide us your "make host_info" output. I'm attaching many here for reference.

[hostinfo.txt](https://github.com/user-attachments/files/21125004/hostinfo.txt)

comejv, 2025-07-08:
Currently I'm using cmake to build nuttx and other firmwares at the same time, like this:
```cmake
ExternalProject_Add(
  nuttx_firmware
  SOURCE_DIR "${CMAKE_CURRENT_SOURCE_DIR}/external/nuttx"
  CONFIGURE_COMMAND
    ${CMAKE_COMMAND}
    -S "${CMAKE_CURRENT_SOURCE_DIR}/external/nuttx"
    -B "${CMAKE_CURRENT_BINARY_DIR}/nuttx_build"
    -G "Ninja"
    -DBOARD_CONFIG=nucleo-h743zi:nsh
    -DNUTTX_APPS_DIR=${CMAKE_CURRENT_SOURCE_DIR}/external/nuttx-apps
  BUILD_COMMAND
    ${CMAKE_COMMAND} --build "${CMAKE_CURRENT_BINARY_DIR}/nuttx_build" -j
    ${CMAKE_HOST_SYSTEM_PROCESSOR_COUNT}
  INSTALL_COMMAND
    ${CMAKE_COMMAND} -E copy
    "${CMAKE_CURRENT_BINARY_DIR}/nuttx_build/nuttx"
    "${OUTPUT_DIR}/nuttx_engine_steering.elf"
    COMMAND
    ${CMAKE_COMMAND} -E copy
    "${CMAKE_CURRENT_BINARY_DIR}/nuttx_build/nuttx.bin"
    "${OUTPUT_DIR}/nuttx_engine_steering.bin"
  BINARY_DIR "${CMAKE_CURRENT_BINARY_DIR}/nuttx_build"
)
```

I just tried compiling only nuttx inside the source directory with both cmake and tools/configure make and neither changed the outcome.
I could not obtain host_info with cmake so I ran `./tools/configure.sh -l nucleo-h743zi:nsh && make host_info` to get it.

[hostinfo.txt](https://github.com/user-attachments/files/21125859/hostinfo.txt)

I'm not sure what you mean by out of tree custom board, when I specify `-DBOARD_CONFIG=nucleo-h743zi:nsh` in cmake configure step doesn't it use the default board config included with nuttx? The only custom things I have are custom apps that I symlink into nuttx-apps/external.

acassis, 2025-07-08:
BTW, this is out of tree usage:
https://nuttx.apache.org/docs/latest/guides/building_nuttx_with_app_out_of_src_tree.html

I think you don't have BASE_DEFCONFIG because you are using CMakefile, I will open an issue about it.

comejv, 2025-07-08:
Regarding out of tree building I see no mentions of CMake, is CMake a recent addition to NuttX?
What I chose to do was a symlink to nuttx-apps/external using a cmake file on which the main target depends:
```cmake
# cmake/symlink_apps.cmake
# This script creates a symbolic link for the main 'apps' directory into the nuttx-apps 'external' directory.

# Define source and destination directories relative to the project root
set(APPS_SOURCE_DIR "${CMAKE_CURRENT_SOURCE_DIR}/nuttx-apps")
set(NUTTX_APPS_DEST_DIR "${CMAKE_CURRENT_SOURCE_DIR}/external/nuttx-apps")

# Define the destination for the symlink
set(SYMLINK_DESTINATION "${NUTTX_APPS_DEST_DIR}/external")

# Remove existing symlink or directory in the destination
if(EXISTS "${SYMLINK_DESTINATION}")
    message(STATUS "Removing existing ${SYMLINK_DESTINATION}")
    file(REMOVE_RECURSE "${SYMLINK_DESTINATION}")
endif()

# Calculate the relative path for the symlink to use relative paths
file(RELATIVE_PATH RELATIVE_APPS_PATH "${NUTTX_APPS_DEST_DIR}" "${APPS_SOURCE_DIR}")

# Create symbolic link from nuttx-apps/external to the main apps directory
execute_process(
    COMMAND ${CMAKE_COMMAND} -E create_symlink
            "${RELATIVE_APPS_PATH}"
            "${SYMLINK_DESTINATION}"
    RESULT_VARIABLE SYMLINK_RESULT
    OUTPUT_VARIABLE SYMLINK_OUTPUT
    ERROR_VARIABLE SYMLINK_ERROR
)

if(NOT SYMLINK_RESULT EQUAL 0)
    message(FATAL_ERROR "Failed to create symlink for apps directory: ${SYMLINK_ERROR}")
else()
    message(STATUS "Created symlink: ${SYMLINK_DESTINATION} -> ${RELATIVE_APPS_PATH}")
endif()
```

I don't remember where I learned about nuttx-apps/external but not in the docs you mentioned.

For anyone wondering if using cmake you do no need a Makefile or Make.defs, only a CMakeLists at the root of external and one in each project:
```cmake
# external/CMakeLists.txt

nuttx_add_subdirectory()
nuttx_generate_kconfig(MENUDESC "*** Custom Firmware Applications ***")
```
```cmake
# external/hello/CMakeLists.txt
if(CONFIG_EXTERNAL_HELLO)
  nuttx_add_application(
    NAME
    hello
    SRCS
    hello_main.c
    STACKSIZE
    1024
    PRIORITY
    100
endif()
```

acassis, 2025-07-08:
@comejv sounds good!

Suggestion, add "-pipe" to your board scripts/Make.defs, just like in my board:

```
weact-stm32h743/scripts/Make.defs:CXXFLAGS := $(ARCHCXXFLAGS) $(ARCHOPTIMIZATION) $(ARCHCPUFLAGS) $(ARCHXXINCLUDES) $(ARCHDEFINES) $(EXTRAFLAGS) -pipe
```

comejv, 2025-07-08:
Adding `-pipe` did not prevent the crash, I checked and it did appear in host_info.

comejv, 2025-07-08:
That's what I did, there was no negation there! The flag was present in host_info.

comejv, 2025-07-08:
With `gcc-arm-11.2-2022.02-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc` nuttx still goes into "Assertion failed panic" after running `ps` two times with `usleep` running in background.

comejv, 2025-07-08:
No differences with GCC 13.2.1 either.

acassis, 2025-07-08:
@comejv really strange, why it works here and doesn't work to you. Although my board is different, the MCU is basically the same.

comejv, 2025-07-09:
I was indeed missing `CONFIG_NSH_ARCHINIT=y` and `CONFIG_BOARD_LATE_INITIALIZE=y`! This plus using `CONFIG_INIT_STACKSIZE` solved the issue. Thanks a lot!

Do you know why I could be missing these? Should they be on by default?

Also do you know any resources for config managment? With my out of tree build rn when saving the config it ends up in `build/nuttx_build/.config`. I have botched together a cmake target that saves and replaces it if need but it's probably not the optimal way. Ideally I'd like to have multiple configs that I could choose at config time.

comejv, 2025-07-09:
Thanks!

With my out of tree build, I'm not sure how to handle defconfigs though. If I put the defconfig in nuttx's build directory where the .config is and I run olddefconfig, it cannot find the Kconfig files.

comejv, 2025-07-10:
I figured it out! From my tests the best way to do it is to create new configs in the nuttx sources, with CMake targets In the end here are the two targets I'm using:
```cmake
add_custom_target(install_custom_defconfig
  COMMAND ${CMAKE_COMMAND} -E make_directory
          "${NUTTX_SOURCE_DIR}/boards/arm/stm32h7/nucleo-h743zi/configs/${NUTTX_CONFIG_NAME}"
  COMMAND ${CMAKE_COMMAND} -E copy_if_different
          "${CMAKE_CURRENT_SOURCE_DIR}/nuttx-configs/${NUTTX_CONFIG_NAME}.defconfig"
          "${NUTTX_SOURCE_DIR}/boards/arm/stm32h7/nucleo-h743zi/configs/${NUTTX_CONFIG_NAME}/defconfig"
  COMMENT "Installing custom defconfig as board configuration: ${NUTTX_CONFIG_NAME}"
)

add_dependencies(install_custom_defconfig symlink_nuttx_apps patch_nuttx_kconfig)

add_custom_target(save_nuttx_config
  COMMAND ${CMAKE_COMMAND} --build "${NUTTX_BUILD_DIR}" --target savedefconfig
  COMMAND ${CMAKE_COMMAND} -E copy_if_different "${NUTTX_BUILD_DIR}/defconfig" "${CMAKE_CURRENT_SOURCE_DIR}/nuttx-configs/${NUTTX_CONFIG_NAME}.defconfig"
  COMMENT "Saving NuttX defconfig to ${CMAKE_CURRENT_SOURCE_DIR}/nuttx-configs/${NUTTX_CONFIG_NAME}.defconfig"
)
```