apache/nuttx #16732: [HELP] Compilation Failure for ADC Example on Nucleo-H743ZI

comejv, 2025-07-15:
### Description

When attempting to compile the `apps/examples/adc` application for the `nucleo-h743zi` board, the build fails with "undeclared" errors for specific ADC GPIO pin definitions (e.g., `GPIO_ADC12_INP5`, `GPIO_ADC123_INP10`, `GPIO_ADC123_INP12`, `GPIO_ADC12_INP13`, `GPIO_ADC12_INP15`) within `boards/arm/stm32h7/nucleo-h743zi/src/stm32_adc.c`:
```c
nuttx/boards/arm/stm32h7/nucleo-h743zi/src/stm32_adc.c:86:3: error: 'GPIO_ADC12_INP5' undeclared here (not in a function); did you mean 'GPIO_ADC12_INP5_0'?
   86 |   GPIO_ADC12_INP5,
      |   ^~~~~~~~~~~~~~~
      |   GPIO_ADC12_INP5_0
[...]
```

When I added these lines to `boards/arm/stm32h7/nucleo-h743zi/include/board.h` it compiled:

```c
/* ADC */

#define GPIO_ADC12_INP5   (GPIO_ADC12_INP5_0 | GPIO_SPEED_50MHz)
#define GPIO_ADC123_INP10 (GPIO_ADC123_INP10_0 | GPIO_SPEED_50MHz)
#define GPIO_ADC123_INP12 (GPIO_ADC123_INP12_0 | GPIO_SPEED_50MHz)
#define GPIO_ADC12_INP13  (GPIO_ADC12_INP13_0 | GPIO_SPEED_50MHz)
#define GPIO_ADC12_INP15  (GPIO_ADC12_INP15_0 | GPIO_SPEED_50MHz)
```

But I'm not seeing any change in values when running `adc` in nsh and sending waveforms in the ADC_IN headers.

My defconfig:
[defconfig.txt](https://github.com/user-attachments/files/21244624/defconfig.txt)

Any idea?

Thanks,
Côme

### Verification

- [x] I have verified before submitting the report.

comejv, 2025-07-17:
I found my issue, it wasn't configuration but using the wrong headers. I was sending voltage to ADC_IN without succes but sending voltage to A1-A5 gives me results!

closed 2025-07-17 by comejv

comejv, 2025-07-18:
I also added this to enable ADC2:
```c
#ifdef CONFIG_STM32H7_ADC2
        /* Configure the pins as analog inputs for the selected channels */

        for (i = 0; i < ADC2_NCHANNELS; i++)
          {
            if (g_adc2_pinlist[i] != 0)
              {
                stm32_configgpio(g_adc2_pinlist[i]);
              }
          }

        /* Call stm32_adcinitialize() to get an instance of the ADC interface */

        adc = stm32h7_adc_initialize(2, g_adc2_chanlist, ADC2_NCHANNELS);
        if (adc == NULL)
          {
            aerr("ERROR: Failed to get ADC2 interface\n");
            return -ENODEV;
          }

        /* Register the ADC driver at "/dev/adc[0-1]" */

        ret = adc_register(devname, adc);
        if (ret < 0)
          {
            aerr("ERROR: adc_register(%s) failed: %d\n", devname, ret);
            return ret;
          }

        devname[8]++;
#endif
```

Is there any reason why ADC2 isn't included in the setup?

raiden00pl, 2025-07-18:
It's impossible to enable all the features of a given dev board in a universal way, especially in the case of highly configurable MCUs such as STM32. Upstream boards should be treated more like a guide or example, not a ready-to-use solution for all applications. In the case of ADCs, you can't configure the ADC pins to suit every user application, as this is highly use case specific. Most likely ADC2 is not included because the contributor who added support for this board apparently didn't use it :)

raiden00pl, 2025-07-22:
Probably yes. If all available ADCs on the chip are enabled in one of the upstream configurations, this will give us better code coverage in our CI tests.