apache/nuttx #16760: [BUG] rand()'s cycle seems too short

yamt, 2025-07-24:
### Description / Steps to reproduce the issue

nuttx's rand() implementation is a bit poor and, with the default
CONFIG_LIBC_RAND_ORDER=1, it seems to produce the same values for
every 903 calls. 903 seems too small to me.

i observed this when playing with mbedtls benchmark.
mbedtls's programs/test/benchmark.c uses rand() for the entropy source.
i guess it's ok because it's just a benchmark.
with its default parameters, mbedtls_mpi_gen_prime happens to consume
231168 bytes random values, which is unfortunately a multiple of sizeof(int) * 903.
it means successive mbedtls_mpi_gen_prime calls always produce identical primes.
it makes mbedtls_rsa_gen_key run forever for its attempt to produce two different
primes, which does never succeed.

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS 15.5

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: OS Components]

### Host information

esp32s3

### Verification

- [x] I have verified before submitting the report.

FelipeMdeO, 2026-08-11:
> @yamt is the issue solved? :-)

Let's close this kind of task. It is stopped for several months and it seems not an issue anymore. 
We can open again if the owner request. 

If we keep it open we may forget return and close it.

cederom, 2026-08-15:
Sure we can close now @FelipeMdeO, thank you, please reopen in case of further issues @yamt :-)

closed 2026-08-15 by cederom