apache/nuttx #16761: [HELP] NuttX fatfs vs elm-chan.org fatfs

Laczen, 2025-07-24:
### Description

NuttX contains support for the fat filesystem. There also is a fatfs library from elm-chan.org that seems to be quite popular. The fatfs library from elm-chan.org seems to be well maintained.

Would there be an advantage to use the fatfs library from elm-chan.org over the NuttX builtin library ?

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-07-24:
@Laczen the NuttX fatfs is very stable and works fine. There are many products using it in the field. But elm-chan.org fatfs has more features including exFAT support (that could require royalties payment to MS). Are you planing to port elm-chan.org to NuttX?

xiaoxiang781216, 2025-07-24:
We are using elm-chan version for many years, here is the porting: https://github.com/open-vela/nuttx/tree/dev/fs/fatfs

Laczen, 2025-07-24:
@acassis @xiaoxiang781216 thanks for replying. I am wondering what (if any) the advantage would be of using elm-chan.org fatfs.

I am mainly interested in creating really small (1-2 MB) disks with maximum compatibility with other systems (windows, linux). If possible I would remotely mount the NuttX disk on either windows or Linux.

@xiaoxiang781216 your porting seems very simple, should this be included upstream?

closed 2026-05-02 by linguini1