apache/nuttx #16762: [HELP] Timer capture on nucleo stm32h743zi

comejv, 2025-07-24:
### Description

Hello, I need to use timer capture on my nucleo-stm32h743zi. I haven't found anything related to it in the sources, but there are lower half drivers for capture in the stm32/ folder (for f boards).

Does that mean timer capture has not been implemented for this architecture?

Would there be a small patch that would enable it?

### Verification

- [x] I have verified before submitting the report.

comejv, 2025-07-30:
To be able to include the source files I just added:
```cmake
    INCLUDE_DIRECTORIES
    ${NUTTX_DIR}/arch/arm/src/stm32h7/)
```
to the CMakeLists of my app.

Using gpio interrupt I seem to be limited by the resolution of the system clock when using `clock_gettime(CLOCK_MONOTONIC, &now)`, which caps me at 100Hz. How can I fix that? Increasing system clock? Going tickless? Using another timer?

acassis, 2025-07-31:
@comejv so you are trying to measure the pulse length using an ordinary GPIO pin instead of using the timer capture? Well, I think the best approach is figuring out how to get the timer capture working, maybe @raiden00pl have found his example. Or you can study esp32 capture driver/app and adapt it to stm32. BTW, if you want to go this way using GPIO, you can use a free running timer to count the time, like I did in the HC-SR04: https://acassis.wordpress.com/2017/08/19/creating-hc-sr04-driver-for-nuttx/

comejv, 2025-07-31:
Hey, thank you I did try a free runner and it's working great, I can measure rising edges up to a MHz. I will probably stick with this because I'm not confident I could port the timer capture driver myself.

Here's how I did it for anyone having the same issue:
[](https://gist.github.com/comejv/271d396c770a2f9a3e478a3f5ea7899a)

I'll be closing this issue but if you have suggestions about how I did it or updates on timer capture on the H7 please tag me!

acassis, 2025-07-31:
@comejv this code is not portable, you are accessing the MCU registers from application side. Also it only will work in FLAT mode, it you try in PROTECTED mode or KERNEL mode it will not work. This is why I suggest you to use the HC-SR04 driver as reference.

raiden00pl, 2025-08-01:
I haven't used TIM capture with NuttX yet, so I can't help. The timers in the stm32 and stm32h7 are basically the same, so the port should be a simple copy-paste from these files:

- arch/arm/src/stm32/stm32_capture.c
- arch/arm/src/stm32/stm32_capture_lowerhalf.c
- boards/arm/stm32/stm32f4discovery/src/stm32_capture.c

comejv, 2025-08-01:
I'm currently looking at porting them rn, not exactly a copy paste but I'm giving it a try. It's mostly defining new macros like in board.h: 
```c
#define GPIO_TIM1_CH1IN   (GPIO_TIM1_CH1IN_1)   /* PE9  (AF1) - D6 */
```
And fixing the driver to use `STM32H7` instead of `STM32`, using APB H and L properly (STM32 doesn't have that).

I've noticed a typo in `external/nuttx/arch/arm/src/stm32f7/stm32_capture.c`:
```c
#define USE_ADVENCED_TIM 1
```

Advance spelled the wrong way. No bug from it because it's consistent in the file. I'll probably do a small PR to fix it next week.

comejv, 2025-08-04:
I'll probably just push the driver as is and drive the timer directly, bypassing the ioctl and lowerhalf driver that are designed for PWM.

acassis, 2025-08-04:
@comejv each PINFUNCTION_n (n = 1, 2, 3, ...) means an alternative pin for a function. For example a GPIO_SPI1_MISO could be in differents pins of the chip, you need to look at arch/arm/src/familyname/hardware/pinmap.h to understand what is the physical pin you want to use to GPIO_SPI1_MISO function. After you understand that logic it is very simple

raiden00pl, 2025-08-05:
@comejv you need a datasheet for a given chip ("Pin descriptions" section). You can find it on the ST website https://www.st.com/en/microcontrollers-microprocessors/stm32h7-series/documentation.html

comejv, 2025-08-05:
I found the right datasheet, thanks. The NuttX pinout has definitions that use specific Alternate Functions, how was the AF number chosen?

For example here we have CH1IN for timer 13, using AF9.
```c
#define GPIO_TIM13_CH1IN_1        (GPIO_ALT|GPIO_AF9|GPIO_FLOAT|GPIO_PORTA|GPIO_PIN6)
```

comejv, 2025-08-05:
@raiden00pl 
> [@comejv](https://github.com/comejv) you need a datasheet for a given chip ("Pin descriptions" section). You can find it on the ST website https://www.st.com/en/microcontrollers-microprocessors/stm32h7-series/documentation.html -> Datasheet -> look at the file name (e.g. https://www.st.com/resource/en/datasheet/stm32h730ab.pdf)

I found the right datasheet, thanks. The NuttX pinout has definitions that use specific Alternate Functions, how was the AF number chosen?

For example here we have CH1IN for timer 13, using AF9.
```c
#define GPIO_TIM13_CH1IN_1        (GPIO_ALT|GPIO_AF9|GPIO_FLOAT|GPIO_PORTA|GPIO_PIN6)
```

comejv, 2025-08-05:
@juniskane I know it's been a really long time (7 years!) but you're the one who authored that part, could you help me understand it?

PR #16809 (2025-08-07): arch/arm/stm32h7: Port timer capture driver from stm32
## Summary

This commit introduces a timer capture driver for the STM32H7 series ported from the STM32 F series.

The current implementation is based on a driver originally for PWM input, and as such, it calculates duty cycle and frequency. It is also limited to a single capture channel per timer.

This serves as a foundation for future development of more comprehensive input capture capabilities on STM32H7 platforms.

The main changes include:
- A new generic timer capture driver for STM32H7.
- A lower-half driver to integrate with the Nu [... cut]
closed 2025-08-07 by acassis