apache/nuttx #16775: [BUG] Conflicting declarations when using toolchain C++

Barto22, 2025-07-25:
### Description / Steps to reproduce the issue

I created a simple C++ application using C++20 features and wanted to use the C++ library from the ARM toolchain. I’m using the latest ARM toolchain version 14.3.1. The target platform is STM32F4Discovery, with the build configuration set to testlibcxx.

In menuconfig, I changed the C++ Library option to "Toolchain C++ support" because the LLVM C++ library produced some errors (which I’ll address as a separate issue).

When compiling, I encounter the following errors:
In file included from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/cstdlib:79,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/ext/string_conversions.h:43,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/bits/basic_string.h:4154,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/string:54,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/bits/locale_classes.h:40,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/locale:41,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/format:47,
                 from ../Documents/nuttx/nuttx-apps/external/main.cpp:3:
../Documents/arm-gnu-toolchain/arm-none-eabi/include/stdlib.h:39:3: error: conflicting declaration 'typedef struct div_t div_t'
   39 | } div_t;
      |   ^~~~~
In file included from ../Documents/nuttx/nuttx/include/nuttx/kmalloc.h:34,
                 from ../Documents/nuttx/nuttx/include/nuttx/lib/lib.h:32,
                 from ../Documents/nuttx/nuttx/include/stdio.h:37,
                 from ../Documents/nuttx/nuttx/include/wchar.h:37,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/cwchar:44,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/bits/postypes.h:40,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/iosfwd:42,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/bits/localefwd.h:41,
                 from ../Documents/arm-gnu-toolchain/arm-none-eabi/include/c++/14.3.1/locale:40:
../Documents/nuttx/nuttx/include/stdlib.h:98:22: note: previous declaration as 'typedef struct div_s div_t'
   98 | typedef struct div_s div_t;
      |                      ^~~~~
../Documents/arm-gnu-toolchain/arm-none-eabi/include/stdlib.h:45:3: error: conflicting declaration 'typedef struct ldiv_t ldiv_t'
   45 | } ldiv_t;
      |   ^~~~~~
../Documents/nuttx/nuttx/include/stdlib.h:108:23: note: previous declaration as 'typedef struct ldiv_s ldiv_t'
  108 | typedef struct ldiv_s ldiv_t;
      |                       ^~~~~~
../Documents/arm-gnu-toolchain/arm-none-eabi/include/stdlib.h:52:3: error: conflicting declaration 'typedef struct lldiv_t lldiv_t'
   52 | } lldiv_t;
      |   ^~~~~~~
../Documents/nuttx/nuttx/include/stdlib.h:118:24: note: previous declaration as 'typedef struct lldiv_s lldiv_t'
  118 | typedef struct lldiv_s lldiv_t;
      |                        ^~~~~~~
../Documents/arm-gnu-toolchain/arm-none-eabi/include/stdlib.h:205:9: error: conflicting declaration of C function 'int putenv(char*)'
  205 | int     putenv (char *__string);
      |         ^~~~~~
../Documents/nuttx/nuttx/include/stdlib.h:162:11: note: previous declaration 'int putenv(const char*)'
  162 | int       putenv(FAR const char *string);
      |           ^~~~~~
ninja: build stopped: subcommand failed.

I am building the project with CMake.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.04.2 LTS

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Build System]

### Verification

- [x] I have verified before submitting the report.

snikeguo, 2025-07-26:
I have also encountered this problem.

If you want to use the C++ provided by the toolchain, you must compile the C++ library of GCC, i.e., libstdc++-v3, by yourself.
I have successfully compiled LLVM's LIBC++ 17.0.6, but it is hardly usable because it occupies a very large stack space. For example:
```
std::vector<u8> tv1;
printf(tv1);
tv1.push_back(1);
```
The above code will use nearly 1 - 2KB of stack space. In fact, when I print the size of tv1, tv1 only occupies dozens of bytes. I tried to decompile the code, as shown in the figure. So LLVM's LIBC is also not available.

[image]

[image]

You can use uclibc++. The above code works fine in uclibc++ and only occupies a very small stack space.

[image]

In conclusion, using C++ is not a very good choice for the time being. Therefore, I have removed the STL from my project entirely.

Barto22, 2025-07-26:
@snikeguo Thanks for your answer! I’ve sent you a private email with a question about some details regarding your integration with NuttX.

acassis, 2025-07-27:
@Barto22 if you can, please keep the discussion public here, this way it could help more people facing this issue in the future. @snikeguo please consider helping to improve the existing Documentation to help more people.

Barto22, 2025-07-28:
I’ve discussed this with @snikeguo, and it seems that only uClibc++ is fully supported. Attempting to include LLVM or a toolchain's C++ standard library (STL) leads to numerous errors, such as “dangerous relocation: unsupported relocation”, as well as various conflicts.

While NuttX’s built-in C libraries and minimal C++ implementation are great for keeping the footprint small, integrating it into projects that depend on the full C++ STL—where footprint is less of a concern—appears to be nearly impossible without significant effort.

Has anyone successfully done this or found a simpler approach to make it work?

xiaoxiang781216, 2025-07-28:
why do you think it need signification work? CI already enable libcxx on many arch/board, you just need follow them:
```
boards/risc-v/mpfs/icicle/configs/knsh/defconfig:61:CONFIG_LIBCXX=y
boards/risc-v/bl602/bl602evb/configs/elf/defconfig:59:CONFIG_LIBCXX=y
boards/risc-v/esp32c3-legacy/esp32c3-devkit/configs/cxx/defconfig:29:CONFIG_LIBCXX=y
boards/risc-v/qemu-rv/rv-virt/configs/citest/defconfig:74:CONFIG_LIBCXX=y
boards/risc-v/qemu-rv/rv-virt/configs/libcxx/defconfig:43:CONFIG_LIBCXX=y
boards/risc-v/qemu-rv/rv-virt/configs/libcxx64/defconfig:43:CONFIG_LIBCXX=y
boards/sim/sim/sim/configs/citest/defconfig:70:CONFIG_LIBCXX=y
boards/sim/sim/sim/configs/matter/defconfig:50:CONFIG_LIBCXX=y
boards/sim/sim/sim/configs/libcxxtest/defconfig:46:CONFIG_LIBCXX=y
boards/sim/sim/sim/configs/tflm/defconfig:34:CONFIG_LIBCXX=y
boards/sim/sim/sim/configs/binder/defconfig:60:CONFIG_LIBCXX=y
boards/xtensa/esp32/esp32-devkitc/configs/cxx/defconfig:31:CONFIG_LIBCXX=y
boards/xtensa/esp32s3/esp32s3-devkit/configs/cxx/defconfig:34:CONFIG_LIBCXX=y
boards/xtensa/esp32s2/esp32s2-saola-1/configs/cxx/defconfig:31:CONFIG_LIBCXX=y
boards/arm/imx6/sabre-6quad/configs/libcxx/defconfig:40:CONFIG_LIBCXX=y
boards/arm/imxrt/imxrt1064-evk/configs/libcxxtest/defconfig:26:CONFIG_LIBCXX=y
boards/arm/imxrt/imxrt1060-evk/configs/libcxxtest/defconfig:26:CONFIG_LIBCXX=y
boards/arm/imxrt/imxrt1170-evk/configs/libcxxtest/defconfig:26:CONFIG_LIBCXX=y
boards/arm/imxrt/imxrt1050-evk/configs/libcxxtest/defconfig:26:CONFIG_LIBCXX=y
boards/arm/stm32/stm32f4discovery/configs/testlibcxx/defconfig:25:CONFIG_LIBCXX=y
boards/arm/qemu/qemu-armv7a/configs/full/defconfig:71:CONFIG_LIBCXX=y
boards/arm/mx8mp/verdin-mx8mp/configs/nsh/defconfig:44:CONFIG_LIBCXX=y
```

Barto22, 2025-07-28:
Did you try to run something from STL like <format>? I tried this configurations for stm32f4discovery and even enable libcxxabi but it have a lot of conflicts and errors which I wrote above.

xiaoxiang781216, 2025-07-28:
Of course, we use many advanced c++ library features(e.g. container, file, thread, closure...).

xiaoxiang781216, 2025-07-28:
But as I said you should avoid to use toolchain prebuilt library.

trns1997, 2025-08-06:
Hey @Barto22 

Here is hopefully a step-by-step guide to build helloxx example which compiles as expected on the stm32f4discovery board (I don’t have hardware to test runtime).
Here’s what I did – hopefully these steps help you get a clean build:

### 1. Update the board configuration

Replace the contents of `boards/arm/stm32/stm32f4discovery/configs/nsh/defconfig` with the following:
```
#
# This file is autogenerated: PLEASE DO NOT EDIT IT.
#
# You can use "make menuconfig" to make any modifications to the installed .config file.
# You can then do "make savedefconfig" to generate a new defconfig file that includes your
# modifications.
#
# CONFIG_ARCH_FPU is not set
# CONFIG_NSH_ARGCAT is not set
# CONFIG_NSH_CMDOPT_HEXDUMP is not set
CONFIG_ARCH="arm"
CONFIG_ARCH_BOARD="stm32f4discovery"
CONFIG_ARCH_BOARD_STM32F4_DISCOVERY=y
CONFIG_ARCH_BUTTONS=y
CONFIG_ARCH_CHIP="stm32"
CONFIG_ARCH_CHIP_STM32=y
CONFIG_ARCH_CHIP_STM32F407VG=y
CONFIG_ARCH_STACKDUMP=y
CONFIG_BOARD_LATE_INITIALIZE=y
CONFIG_BOARD_LOOPSPERMSEC=16717
CONFIG_BUILTIN=y
CONFIG_EXAMPLES_HELLO=y
CONFIG_FS_PROCFS=y
CONFIG_HAVE_CXX=y
CONFIG_HOST_WINDOWS=y
CONFIG_INIT_ENTRYPOINT="nsh_main"
CONFIG_INTELHEX_BINARY=y
CONFIG_LINE_MAX=64
CONFIG_MM_REGIONS=2
CONFIG_NSH_BUILTIN_APPS=y
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_READLINE=y
CONFIG_PREALLOC_TIMERS=4
CONFIG_RAM_SIZE=114688
CONFIG_RAM_START=0x20000000
CONFIG_RAW_BINARY=y
CONFIG_RR_INTERVAL=200
CONFIG_SCHED_WAITPID=y
CONFIG_START_DAY=6
CONFIG_START_MONTH=12
CONFIG_START_YEAR=2011
CONFIG_STM32_JTAG_SW_ENABLE=y
CONFIG_STM32_PWR=y
CONFIG_STM32_SPI1=y
CONFIG_STM32_USART2=y
CONFIG_SYSTEM_NSH=y
CONFIG_USART2_RXBUFSIZE=128
CONFIG_USART2_SERIAL_CONSOLE=y
CONFIG_USART2_TXBUFSIZE=128
CONFIG_CXX_EXCEPTION=y
CONFIG_CXX_RTTI=y
CONFIG_CXX_STANDARD="c++17"
CONFIG_LIBCXXTOOLCHAIN=y
CONFIG_TESTING_CXXTEST=y
CONFIG_EXAMPLES_HELLOXX=y

```
This config enables the build of the helloxx app and links the toolchain builtin library.

### 2. Build commands
```
make distclean
./tools/configure.sh -l stm32f4discovery:nsh
make -j8
```

### 3. Expected output
You should see something similar to:
```
➜  nuttx git:(master) ✗ make -j8
Create version.h
LN: platform/board to /home/thomas/nuttxspace/apps/platform/dummy
Register: hello
Register: dd
Register: nsh
Register: sh
Register: helloxx
Register: cxxtest
CC:  inode/fs_inodebasename.c chip/stm32_gpio.c:44:11: note: '#pragma message: CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py'
   44 | #  pragma message "CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py"
      |           ^~~~~~~
CC:  dirent/lib_opendir.c In file included from chip/stm32_flash.c:40:
chip/stm32f20xxf40xx_flash.c:53:4: warning: #warning "Default Flash Configuration Used - See Override Flash Size Designator" [-Wcpp]
   53 | #  warning "Default Flash Configuration Used - See Override Flash Size Designator"
      |    ^~~~~~~
LD: nuttx
arm-none-eabi-ld: warning: /home/thomas/nuttxspace/nuttx/nuttx has a LOAD segment with RWX permissions
Memory region         Used Size  Region Size  %age Used
           flash:      130872 B         1 MB     12.48%
            sram:        5800 B       112 KB      5.06%
CP: nuttx.hex
CP: nuttx.bin

```

### Notes
Couple notes though:
- I built this on Ubuntu 24.04 with gcc 13.3.0. Since you’re using gcc 14.3.1, some of your issues might be related to that newer toolchain. If possible, try building with gcc 13.x to see if it resolves the conflicts.
- The config above sets C++17 as the standard. If you need C++20 features (e.g., std::format, ranges), you’ll need to adjust and test accordingly — I haven’t validated that yet.

Can you try these exact steps and see if it resolves your build errors?

And maybe @leducp might have other ideas on this matter, not sure if he had the occasion to test out newer gcc versions and recent C++ features on nuttx? If i have the time i might test it out one of these days out of curiosity :) .

leducp, 2025-08-06:
For the record, quick and dirty workaround for the undefined __xpg_error_r symbol:

```cpp
#include <string.h>

extern "C" char * __xpg_strerror_r(int errnum, char * buf, size_t buflen)
{
  (void) strerror_r(errnum, buf, buflen);
  return buf;
}
```

You can put this anywhere in your C++ code, or without the extern "C" in a C file.

Note that this is an improper wrapper: the GNU version of strerror_r (which __xpg_error_r is the underlying implementation) is a bit different of the XSI version provided by NuttX.

trns1997, 2025-08-06:
I just built and tested the following using `CONFIG_CXX_STANDARD="c++20"` on a `XMC4800`:
```
#include <nuttx/config.h>

#include <cstdio>
#include <debug.h>

#include <ranges>
#include <vector>
#include <numeric>

extern "C" int main(int argc, FAR char *argv[])
{
    std::vector<int> v(10);
    std::iota(v.begin(), v.end(), 1); // fills with 1,2,…,10

    auto is_even = [](int x) { return x % 2 == 0; };
    auto square  = [](int x) { return x * x; };

    // Compose a view: filter evens, then square, then take first 3 results
    auto result = v
        | std::views::filter(is_even)
        | std::views::transform(square)
        | std::views::take(3);

    for (int x : result) {
        printf("%d\n", x);
    }

    return 0;
}
```

### Output in picocom
```
nsh> helloxx
4
16
36
```

So views and ranges seem to be working just fine, cannot guarantee that all C++20 features will work but this is super nice to know for all future users :)

Barto22, 2026-01-07:
Master