apache/nuttx #16814: [HELP] <title>add 802.1Q header

ken-hutchinson, 2025-08-06:
### Description

Hello, I need to add priority ethernet messaging. My LAN9370 supports 802.1Q with VLAN. 
I can't figure out where the level 2 packets are put together to add the extra 32 bits to the frame.
Can anyone point me to the correct area of the code?
thanks
Ken

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-08-08:
@xiaoxiang781216 any idea?

acassis, 2025-09-02:
@matiamic since you are working with low level Ethernet 10BASE-T1x, could you please take a look here?

@wengzhe @raiden00pl for visibility

matiamic, 2025-09-03:
Hi @ken-hutchinson, this seems like the default behavior to me: https://man7.org/linux/man-pages/man2/recv.2.html
> If no messages are available at the socket, the receive calls wait for a message to arrive, unless the socket is nonblocking

You should be able to go nonblocking by passing the `MSG_DONTWAIT` flag to the `recv`.

I am not sure about the blocking of the rest of the network. The behavior is defined [here](https://github.com/apache/nuttx/blob/5592dc75dc5fe03b82acf3a0349bd733099e5d60/net/pkt/pkt_recvmsg.c#L430-L489) I believe.
The `net_lock()` is being called, but the lock should be released by the `net_sem_wait` during waiting. But it seems from your description that the netlock might still be locked or there is some other mechanism blocking your ping...

ken-hutchinson, 2025-09-03:
Thanks for your reply @matiamic 
The blocking call is probably fine.  
I have the receiver doing 'read' in a loop with 1m-sec sleep between calls. The sender can send a ping, or raw frames (with 802.1q tag). If I ping first, they go fine. The raw frame is fine. The first 'ping' after a raw frame always fails. If I ping again, it works.  Does this mean anything to you? 
Could also be a problem with the vlan config on the LAN9370.  On the failed pings, I can see in the LAN9370 stats the data enters the RGMII MAC port, but doesn't make it to the outgoing port. Don't know why that would happen.
thanks
Ken

matiamic, 2025-09-04:
Sounds like expected behavior of [`arp_out`](https://github.com/apache/nuttx/blob/b5f6620a4c85735b28bbfdf2a806f4f40ea4de95/net/arp/arp_out.c#L106-L133).

This should happen only after reset (or when arp cache is dropped maybe). The incoming ICMP echo request is processed by the net stack and the ICMP echo reply is then saved into the output buffer. Before sending, the `arp_out` is called. No arp cache entry is found for the given IP, the contents of the output buffer are replaced with the ARP request, the ICMP reply is lost forever.

acassis, 2025-09-04:
Hi @ken-hutchinson, is it possible to you share your code? This way more people could analyze and help to test. I see that NuttX has some support to 802.1Q, but there is no documentation about it. Probably someone used, got it working on their board and never submitted it to mainline.

I tried to find some development board with LAN9370 to test, but I didn't find any. Do you know some chip that could be used to test it?