apache/nuttx #16973: [BUG] Non-blocking make export error: missing gnu-elf.ld

trns1997, 2025-09-03:
### Description / Steps to reproduce the issue

When I configure for `stm32f4discovery:nsh` and then run `make export`, I get a warning about a missing linker script:
```
cp: cannot stat '/home/thomas/nuttxspace/nuttx/libs/libc/elf/gnu-elf.ld': No such file or directory
```

Despite this, the export archive is created successfully and the build is not blocked. It seems the export step is trying to copy a file that doesn’t exist anymore.

### Steps to Reproduce:

1. Configure NuttX:
```
./tools/configure.sh -l stm32f4discovery:nsh
```
2. Run export:
```
make export
```

**Output**:
```
Create version.h
LN: platform/board to /home/thomas/nuttxspace/apps/platform/dummy
Register: hello
Register: dd
Register: nsh
Register: sh
CC:  chip/stm32_gpio.c chip/stm32_gpio.c:44:11: note: '#pragma message: CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py'
   44 | #  pragma message "CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py"
      |           ^~~~~~~
CC:  chip/stm32_flash.c In file included from chip/stm32_flash.c:40:
chip/stm32f20xxf40xx_flash.c:53:4: warning: #warning "Default Flash Configuration Used - See Override Flash Size Designator" [-Wcpp]
   53 | #  warning "Default Flash Configuration Used - See Override Flash Size Designator"
      |    ^~~~~~~
cp: cannot stat '/home/thomas/nuttxspace/nuttx/libs/libc/elf/gnu-elf.ld': No such file or directory
```

*Notes*:
The error is non-blocking and does not prevent the export archive from being created. I just wanted to check if this is an intentional leftover or something that should be cleaned up. I’d be happy to look into it further if needed.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.04

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Configuring]

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-09-04:
@leducp did you get that same issue on your board?

trns1997, 2025-09-04:
> [@leducp](https://github.com/leducp) did you get that same issue on your board?

@acassis it seems to occur on all boards, i tried with the xmc4800 and i have the same error when i export

trns1997, 2025-09-04:
This bug started occurring since `nuttx-12.9.0`, all releases before this version show no sign of failing `cp`

trns1997, 2025-09-04:
So `mkexport.sh`. The script still assumes that `gnu-elf.ld` always exists, but that file was removed and replaced with a template `gnu-elf.ld.in`. The actual `gnu-elf.ld` is only generated when `CONFIG_LIBC_MODLIB=y` (via modlib).

So during `make export`, if `CONFIG_LIBC_MODLIB=n`, the file never gets generated, and the script tries to copy something that doesn’t exist. That explains the warning we see.

A possible solution might be to update mkexport.sh to check for both cases:
```
APPLD=gnu-elf.ld
APPLDIN=gnu-elf.ld.in

if [ -f "${BOARDDIR}/scripts/${APPLD}" ]; then
  cp -f "${BOARDDIR}/scripts/${APPLD}" "${EXPORTDIR}/scripts/."
elif [ -f "${TOPDIR}/libs/libc/elf/${APPLD}" ]; then
  cp -f "${TOPDIR}/libs/libc/elf/${APPLD}" "${EXPORTDIR}/scripts/."
elif [ -f "${TOPDIR}/libs/libc/elf/${APPLDIN}" ]; then
  # Fall back to exporting the template if the real file is missing
  cp -f "${TOPDIR}/libs/libc/elf/${APPLDIN}" "${EXPORTDIR}/scripts/${APPLD}"
fi
```

This way:

If a board-specific or pre-generated `gnu-elf.ld` exists, it’s copied as before.

If only the .in template exists, it’s exported under the expected name so the export step doesn’t fail.

An alternative would be to trigger the generation of `gnu-elf.ld` from `gnu-elf.ld.in` inside `mkexport.sh` (when CONFIG_LIBC_MODLIB=y), but that feels heavier and could complicate the export flow. Falling back to the template seems like the safest minimal change.

acassis, 2025-09-04:
@trns1997 nice catch. But I think you shouldn't force the generation of this gnu-elf.ld since CONFIG_LIBC_MODLIB is not enabled and probably some program will not be able to use it later.

The propser solution should be verifying if LIBC_MODLIB is enabled and only try to copy when it is enabled.

@anchao @leducp @yf13 comments?