apache/nuttx #17101: [HELP] Does Nuttx have a full-duplex ethernet driver?

GraceYuRPI, 2025-09-26:
### Description

Does Nuttx support full-duplex ethernet? I am looking for a list of drivers needed to create ethernet communication between a few STM32 H753ZI and a Pixhawk 6x (same STM32 H7 chip) using Nuttx. The goal is to use the standard double twisted-pairs ethernet protocol with an unmanaged switch. 

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-09-27:
yes, it does (FULL Duplex support is more about PHY support than NuttX by itself).

pkarashchenko, 2025-09-27:
Yes. Half or full duplex is related to Ethernet PHY configuration and/or auto-negotiation process, so you will need to check your PHY driver.